arduino-libraries/WiFi101 #129: WifiClient post fails after 6 attempts

nsollars, 2016-12-31:
Hi all,

I am not sure if this issue has been covered before,  im using  simple http post to neo4j using basic auth and content set to json.

like I said I get 6 good tones then it fails to connect ever again ... ive tried just about everything to solve this one  but to no avail ...

Just for more clarity here is the good responses to the serial / console..

Making Post Attempt
HTTP/1.1 200 OK
Date: Sat, 31 Dec 2016 02:37:54 GMT
Content-Type: application/json
Access-Control-Allow-Origin: *
Connection: close
Server: Jetty(9.2.9.v20150224)

{"results":[{"columns":["gtcu","sensor"],"data":[{"row":[{"Uniqueid":66345,"IMSI":763984509326745,"IMEI":345876456374960,"Firmware":"21.104","SerialID":"CNSECM1424000006"},{"SensorID":72008,"Type":"Door_Sensor","Hardware_Rev":"1.0","Software_Rev":"1.0"}],"meta":[{"id":13,"type":"node","deleted":false},{"id":27,"type":"node","deleted":false}]},{"row":[{"Uniqueid":66345,"IMSI":763984509326745,"IMEI":345876456374960,"Firmware":"21.104","SerialID":"CNSECM1424000006"},{"SensorID":72015,"Type":"Temperature_Sensor","Hardware_Rev":"1.5","Software_Rev":"1.1"}],"meta":[{"id":13,"type":"node","deleted":false},{"id":24,"type":"node","deleted":false}]},{"row":[{"Uniqueid":66345,"IMSI":763984509326745,"IMEI":345876456374960,"Firmware":"21.104","SerialID":"CNSECM1424000006"},{"SensorID":72022,"Type":"Temperature_Sensor","Hardware_Rev":"1.5","Software_Rev":"1.1"}],"meta":[{"id":13,"type":"node","deleted":false},{"id":20,"type":"node","deleted":false}]}]}],"errors":[]}

disconnecting from server.

after 6 go arounds I just keep getting ... 

Making Post Attempt
No we failed to connect to server this time

disconnecting from server.

I am wondering if perhaps the client.stop() call is not clearing down the needful ... I am seeing two "Making Post Attempt" entries before getting the return data,  so also could be my timing is way off too.

I was using a modified version of the simplewebclient sketch to get this done, basically just header changes and the http post code in the loop() section vs the setup() section like the example...

Any help would be appreciated,

Thanks in advance

Nige

[sketch_code.txt](https://github.com/arduino-libraries/WiFi101/files/679440/sketch_code.txt)

sandeepmistry, 2017-01-03:
Hi @nsollars,

From looking at your sketch I think the problem is that your are not closing the socket connection after reading the response body in the successful path.

Could you please try changing the following lines from:
```
wifiClient.flush();
```

to

```
wifiClient.flush();
wifiClient.stop(); // disconnect the socket
```

sandeepmistry, 2017-01-19:
Closing this for now due to lack of feedback.

closed 2017-01-19 by sandeepmistry