arduino-libraries/WiFi101 #130: Server socket failure after disconnect from AP

driverblock, 2017-01-01:
This seems similar to #46, but not the same.  Attached code creates a telnet server. First-time connections are successful.  After receiving a 'D' character from the client, the server will close the connection and disconnect from the access point.  After a short delay, it reconnects to the access point, re-initializes the server, and starts listening again.
Subsequent connections report successful on the client side, but client.available() appears to hang on the server side.
Code is written for and tested on Adafruit Feather M0.  Sorry it's a bit messy - it's based on forum user code and modified for debugging purposes.
[AP_disconnect_testcase.zip](https://github.com/arduino-libraries/WiFi101/files/680045/AP_disconnect_testcase.zip)

sandeepmistry, 2017-01-03:
@driverblock this seems ok to me with the latest Github master version.

Could you please try this version? There have been a few fixes in since the v0.11.2 release ...

driverblock, 2017-01-03:
Yes, it works with the latest master version.
Thank you for your time!

sandeepmistry, 2017-01-03:
@driverblock excellent, I'll close this issue now then.

A future WiFi101 library release will be include the latest changes from master.

closed 2017-01-03 by sandeepmistry

sandeepmistry, 2017-01-05:
v0.12.0 has been released today with the latest changes from master.