arduino-libraries/WiFi101 #131: Feature request: mDNS / DNS-SD (Bonjour / Zeroconf)

toqduj, 2017-01-01:
It would be very useful for the IoT (in particular for the MKR1000) if we had options for advertising services on the MKR1000, and/or finding advertised services on local networks. I understood this requires detailed access to the WiFi chip functionality, which Wifi101 might have. 
Example for ethernet: 
https://github.com/TrippyLighting/EthernetBonjour

sandeepmistry, 2017-04-25:
Hi @toqduj,

We've created a fork of EthernetBonjour which supports both Ethernet and WiFi101 capable boards: https://github.com/arduino-libraries/ArduinoMDNS

Please let us know if you have any feedback before we make an initial release for the Library Manager.

closed 2017-04-25 by sandeepmistry

toqduj, 2017-04-28:
Awesome, thanks!

toqduj, 2017-05-04:
Awesome, I'll have to give it a try!

sandeepmistry, 2017-05-10:
@toqduj great, the library is also now available in the IDE's Library Manager, just searcher for ArduinoMDNS.

toqduj, 2018-01-06:
I don't see it in the library manager...

toqduj, 2018-01-07:
... any chance it'll be merged back with the TrippyLighting branch? They've got one more fix that might be important...

sandeepmistry, 2018-01-08:
@toqduj could you please open an issue here for this: https://github.com/arduino-libraries/ArduinoMDNS

We've enabled the issue tracker. What specific fix are you referring to?