arduino-libraries/WiFi101 #133: SimpleWebServerWiFi hangs after manually refresh the page 3-5 times

shawhu, 2017-01-13:
Guys,

I've tried the following combination, all failed. I've the latest version of IDE 1.8.0 on a win10x64 machine, I've installed the latest wifi101 library 0.12 I believe.

The problem can be easily reproduced by attaching a Wifi101 shield on top of a Uno board, running out-of-box SimpleWebServerWiFi.ino, only changed the ssid and the password in order to connect.

When first load the page in firefox, most of the time it's successful, and then if you refresh the page or click the link in order to turn the LED on and off for couple times it hangs. and when it hung it can't return to normal unless you reset the board. The serial monitor log I pasted below was produced with 3 attempts, the first one shows the page successfully. the 2nd and the 3rd were not.

GET / HTTP/1.1
Host: 192.168.17.111
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; rv:47.0) Gecko/20100101 Firefox/47.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Connection: keep-alive

client disonnected
new client
client disonnected
new client

I tried to find a solution and I read this #8 
It seems like a bug that's been fixed long time ago. But the characteristics are very similar from what I've 
observed.

please help.

P.S. I've tried with a mega board as well, same result.

Sembazuru, 2017-02-14:
I've seen something similar, especially if the served page has an auto-refresh header. It does work with Chrome. So that might point to something...

sandeepmistry, 2017-02-16:
@shawhu what do you see if the browser when things are "broken"? How quickly are you clicking the buttons and refreshing the page?

This might be related to https://github.com/arduino-libraries/WiFi101/issues/36.

feikname, 2017-05-07:
I can confirm that this happened to me once.

sandeepmistry, 2017-05-10:
@feikname what firmware version were you using? Do you have a specific set of steps that can reproduce the issue?

feikname, 2017-05-10:
@sandeepmistry I will be able to test this again in a few days (hopefully). It was the latest firmware version available for model B.

sandeepmistry, 2017-05-24:
@feikname @shawhu any updates on this issue?

feikname, 2017-05-25:
@sandeepmistry

## Environment
 * Arduino IDE version: 1.8.1 (on Windows 10)
 * Arduino IDE SAMD boards package version: 1.6.15
 * WiFi101 library version:  0.14.2
 * MKR100 WiFi101 firmware version: 19.5.2 (via `CheckWifi101FirmwareVersion.ino` example sketch included in the library)

## Chrome 58.0.3029.110 (64-bits) on Windows 10
`SimpleWifiWebServer` example ran just fine, I manually refreshed the page by spamming the F5 at different frequencies and encountered no problems. `WiFiWebServer` example ran fine too. The custom code (at the bottom of this comment ) did ran fine too.

## Firefox Nightly 55.0a1 (64-bits) on Windows
I only tested the custom code at the bottom of this comment with Firefox, but the results should be applicable to the other examples too.

Nightly received the content just fine for the first time, but then it was unable to make any subsequent requests.

Relevant output:
```text
signal strength (RSSI):-72 dBm
new client connected ← /, probably
client disconnected
new client connected ← favicon, probably
client disconnected
new client connected ← never closed?!
```

What happens if `client.println()` fails somehow? Does it keep in a eternal loop trying to send the message? Perhaps that's where the code got stuck.

After I reset the MKR1000 and ran the test on Nightly again, it worked just fine. I recompiled the code to see the browser request message, and it stopped asking for the favicon but would send out *totally empty* responses instead (?!). 

I could reproduce this issue with Nightly again by changing the port from 80 to something else, but that stopped working after a while and the bug didn't occur again after that. (what a nasty one)

## Custom code
```arduino
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = ""; // your network SSID (name)
char pass[] = ""; // your network password

int status = WL_IDLE_STATUS;

WiFiServer server(80);

void setup() {
  Serial.begin(9600);
  while(!Serial);

  if(WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    while(true); // Halt
  }

  while(status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    
    status = WiFi.begin(ssid, pass);

    delay(4000);
  }
  
  server.begin();
  
  printWiFiStatus();
}

void loop() {
  // listen for incoming clients
  WiFiClient client = server.available();
  
  if(client) {
    Serial.println("new client connected");

    boolean currentLineIsBlank = true;
    while(client.connected()) {
      if(client.available()) {
        char c = client.read();
        // Serial.print(c);

        if (c == '\n' && currentLineIsBlank) {
          // send a long response, for testing purposes (WARNING: NOT VALID HTML)
          for(int i=0; i<50; i++) {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-Type: text/html");
            client.println("Connection: close");  // the connection will be closed after completion of the response
            client.println();
            client.println("<!DOCTYPE HTML>");
            client.println("<html>");
            client.println("millis(): " + String(millis()));
            client.println("</html>" + String(i));
          }
          
          break; 
        }
        
        if(c == '\n') {
          currentLineIsBlank = true;
        }
        else if(c != '\r') {
          currentLineIsBlank = false;
        }
      }
    }
    
    //delay(1);

    client.stop();
    Serial.println("client disconnected");
  }
}

void printWiFiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

**Note:** Is `delay(1)` really necessary? I had not issued transferring data that took way more than one second without it.

shawhu, 2017-05-25:
I'm back. I'll be able to do tests start from tomorrow morning. tell me what do you need to reproduce it.

and yes, this only happens if you refresh the page. I wouldn't say I hit the refresh button to the death but interval like once/5 seconds is more than enough to reproduce it.

sandeepmistry, 2017-05-25:
Hi @feikname,

Thanks for the info!

Unfortunately I tried the steps you provided for Firefox on macOS and a Windows 10 VM, but couldn't reproduce. I've also made PR #173 to speed up `client.println(...) and `client.write(...)` if a client is disconnected - not sure it will help you, but it would be awesome if you can try it out.

sandeepmistry, 2017-05-25:
@shawhu it would be great if you can try this again after updating to f/w 19.5.2: https://www.arduino.cc/en/Tutorial/FirmwareUpdater

Then let us know of the results. Thanks.

feikname, 2017-05-25:
@sandeepmistry Could you try on a (non cached) port on Firefox in Windows 10 VM again? Perhaps it might work this time. I believe this is a case of a [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug) 😟.

shawhu, 2017-05-28:
@sandeepmistry I've tried the updater, my wifi101 is model A, and I don't think I can upgrade it to use f/w 19.5.2, I ran the check101firmwareversion and it says the 19.4.4 version is my latest one.

sandeepmistry, 2017-05-29:
Hey @feikname,

Thanks for the tip to try another port with Windows 10, I was able to reproduce it! It looks like PR #173 solves the issue, so I'll get someone here to review it.

sandeepmistry, 2017-05-29:
@shawhu unfortunately Microchip/Atmel has decided not to provide firmware updates for model A chips :( so 19.4.4 is the latest release available.

feikname, 2017-05-29:
@sandeepmistr

What is the difference between these two WiFi101 firmware versions by the way? I couldn't find any changelog.

sandeepmistry, 2017-05-30:
@feikname here's a copy of the release notes: [Atmel_WINC1500_Software_Release_Notes.pdf](https://github.com/arduino-libraries/WiFi101/files/1038288/Atmel_WINC1500_Software_Release_Notes.pdf)

sandeepmistry, 2017-06-01:
@feikname PR #173 is included in the latest v0.14.3 release that will be on the library manager shortly.

feikname, 2017-06-01:
@sandeepmistry thanks!

sandeepmistry, 2017-06-02:
@shawhu could you please try the new v0.14.3 WiFi101 library release to see if the issue occurs.

If it does occur could you also try to ping the IP address of the board. A Wireshark capture of all the traffic would be great as well!

sandeepmistry, 2017-10-18:
Closing for now due to lack of feedback.

closed 2017-10-18 by sandeepmistry