arduino-libraries/WiFi101 #134: WiFi.getTime() returns tomorrow's day

bcraigie, 2017-01-13:
WiFi101 version 0.12.0 seems to have a bug at line 1005 in WiFi.cpp

It is adding 1 to the day incorrectly, so that today (13th Jan 2017) actually becomes 14th.

Line 1005 reads:
`		tm.tm_mday = systemTime.u8Day + 1;`
Removing the +1 fixes the problem.

The tm struct definition says the day ranges from 1 to 31 so there's no reason to add 1.

drp0, 2017-01-14:
I agree.
I also suggest increasing the while loop wait time at line 994 to 15 seconds:
while (_resolve != 0 && millis() - start < 15000) {

David

closed 2017-01-19 by sandeepmistry

drp0, 2017-01-20:
Have you also increased the wait time to 15s?
I experienced repeated failure at 5s due to latency and (presumeably) high time server demand.
David

sandeepmistry, 2017-01-20:
@drp0 please open a separate issue with a minimal sketch that we can use to reproduce this condition.

PR #136 (2017-01-19): Removed +1 from day
Fixes #134 and http://forum.arduino.cc/index.php?topic=442723.0
commit 3cba4817 (2017-01-19): Removed +1 from day

Fixes #134 and http://forum.arduino.cc/index.php?topic=442723.0