arduino-libraries/WiFi101 #137: Epoch time returning 0 because 5 second timeout too short due to slow network response

drp0, 2017-01-21:
reference [https://github.com/arduino-libraries/WiFi101/issues/134#issuecomment-274087247](url)

Regarding the epoch time request,

The WiFi.getTime() timeout is 5000mS.
This is controlled by line 994 in WiFi.cpp
"while (_resolve != 0 && millis() - start < 5000) {"

My network repeatedly returns zero in the library 5 second period.
Increasing the timeout to 15000ms removes the issue.
Test code:
```C++
// TimetestBasic.ino

// D.R.Patterson
// 21/1/2017

// Tested on a AtMega 2560 

#include <time.h>
#include <SPI.h>
#include <WiFi101.h>

#define MY_TIME_ZONE 0 * ONE_HOUR           // gmt offset 0

const byte wifiSelect = 10; // YiFi101 select (active low)
const byte sdSelect = 53;   // SD select (active low)
int mystatus = WL_IDLE_STATUS;
const char ssid[] = "xxxx";  //  your network SSID (name)

void setup(){
unsigned int date;
time_t tnow;
struct tm lt;
set_zone(MY_TIME_ZONE); // GMT 0 in my case
unsigned long epoch;

// setup wifi and sd select
pinMode(sdSelect, OUTPUT); digitalWrite(sdSelect, HIGH);
pinMode(wifiSelect, OUTPUT); digitalWrite(wifiSelect, LOW);

Serial.begin(115200);
  while(!Serial) yield();
Serial.println("Connecting to Router");
router();

  for(byte i = 0 ; i < 10; i++){
    if (i > 0) delay(5000);
  epoch = WiFi.getTime(); // get wifi time
  Serial.print("Epoch "); Serial.println(epoch);
    if(epoch == 0)continue;
  set_system_time( (time_t)epoch );     // set the time library

  time(&tnow); // store time in variable tnow
  localtime_r(&tnow, &lt);       // place local time in lt
  date = lt.tm_year + 1900;
    if(date < 2021) break;
  }
  
  if ( (date > 2020) || (epoch == 0)) Serial.println("Wifi101 failed to return a valid time");
  else{
  Serial.println("Clock set using WiFi.getTime()");
  Serial.print(isotime(&lt)); // The isotime function constructs an ascii string from the time code
  }
}

void loop(){
}

void router(){
// check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
  Serial.println("WiFi shield not present");
    while(1);
  }else{
  // attempt to connect to Wifi network:
  byte attempts = 0;  
    while ( attempts < 3) {
    attempts++;
    Serial.print("Attempting to connect to SSID: "); Serial.println(ssid);
    mystatus = WiFi.begin(ssid);
    unsigned long tt = millis();
      while ((millis()-tt) < 10000){ // up to 10 second wait
      delay(1000);
      mystatus = WiFi.status();
        if ((mystatus == WL_CONNECTED) || (mystatus == WL_NO_SHIELD)) break;
      }
      if ((mystatus == WL_CONNECTED) || (mystatus == WL_NO_SHIELD)) break;
    }
  }

  if (mystatus == WL_CONNECTED) Serial.println("Wifi Connected\n");
  else{
  Serial.println("Wifi not available\n");
    while(1);
  }
}

```

sandeepmistry, 2017-01-24:
@drp0 this seems ok on my Uno + WiFi101 shield.

Here's then output from the serial monitor:

```
Connecting to Router
Attempting to connect to SSID: F0645C
Wifi Connected

24
Epoch 538586784
Clock set using WiFi.getTime()
2017-01-24 15:26:24
```

What should I expect?

drp0, 2017-01-24:
I am in the UK.
I have not checked, but is the time server  used by the library in the usa? 
When I run the test software I gave you, I get repeated epoch zeros, which may, or may not, produce a non zero value before the 10 attempts are up.
With the library timeout extended I get a non zero epoch value returned immediately.

On my pc. When I manally start an internet time request for the clock in windows, there is a similay delay from American time servers.
David

sandeepmistry, 2017-01-24:
From analyzing the WINC1500 firmware, I would say the time servers are in the US:

```
time-c.nist.gov
time-d.nist.gov
wolfnisttime.com
```

I'm based in Canada, but will ask one of my co-workers to try the sketch you provided. @agdl would you mind trying this?

agdl, 2017-01-25:
It works in Italy too! response time is less tan 1 second

sandeepmistry, 2017-02-16:
@drp0 to help us better understand what's going on, could you please add the following statement:

```
Serial.println(millis());
```

here: https://github.com/arduino-libraries/WiFi101/blob/master/src/WiFi.cpp#L193

Then run your sketch again and provide us the output from the Serial monitor.

Thanks.

drp0, 2017-02-17:
Hi Sandeep,
Do you want millis() prior to the if, or before memcpy (line 194):
		if (WiFi._resolve != 0) {
				memcpy((tstrSystemTime *)WiFi._resolve, pvMsg, sizeof(tstrSystemTime));

				WiFi._resolve = 0;
}

David

sandeepmistry, 2017-02-17:
@drp0 before the `if (WiFi._resolve != 0) {` line please.

drp0, 2017-02-24:
Hi Sandeep,
it looks like the timeout is 2-3 millis seconds and therfore unlikely to be related to the 5 seconds specified in line 994.
My function delays 5000 ms between re-tries

Connecting to Router
Attempting to connect to SSID: DRP3
Wifi Connected

2608
Epoch 0
7610
Epoch 0
12613
Epoch 0
17616
Epoch 0
22619
Epoch 0
27622
Epoch 0
32624
Epoch 541259064
Clock set using WiFi.getTime()
2017-02-24 13:44:24

David

sandeepmistry, 2017-02-24:
@drp0 so I think this means it just takes your WINC1500 ~30 seconds to fetch the NTP time on start.

sandeepmistry, 2017-03-13:
I'm closing this, things are behaving as I expect.

closed 2017-03-13 by sandeepmistry