arduino-libraries/WiFi101 #138: Buffer overflow for TCP packets > MTU size

delcomp, 2017-01-26:
I'm using a MKR1000 with the latest WiFi101 library and IDE 1.8.0 and try to communicatie with a server (SSL / HTTPS) which responds with JSON structured data. This all works will unless the server sends packages larger then the MTU size. Then 10-20 bytes get lost, which of course corrupts the JSON data.

I've written tons of test code to pin point this problem.
Final testcode is below (don't mind the structure, it's no good, just for testing):
        while (https.available() || https.connected()) {
          if (https.available()) {  
            while ((j = https.read()) >= 0) if (i<(MAX_RESPONSE-1)) sResponse[i++] = (char)j;
          }
        }

The HTTP data for a package of appr. 1700 bytes:

Response: Server: nginx
Date: Thu, 26 Jan 2017 19:08:10 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 1207
Connection: close
Status: 200 OK
X-Frame-Options: SAMEORIGIN
X-XSS-Protection: 1; mode=block
X-Content-Type-Options: nosniff
X-TXID: 5b77aa91c941ec5e0149acd23b8f5cec
ETag: W/"aecb81b03d7e634e82bbab87d7471b4c"
Cache-Control: max-age=0, private, must-revalidate
X-Request-Id: 7b806279-c409-435f-b22e-f4b8a30019f6
X-Runtime: 0.654347
X-Frame-Options: SAMEORIGIN

{"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":null,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"<invalid>","battery_range":91.45,"est_battery_range":55.09,"ideal_battery_range":73.16,"battery_level":31,"usable_battery_level":30,"battery_current":0.0,"charge_energy_added":35.54,"charge_miles_added_rated":138.0,"charge_miles_added_ideal":110.5,"charger_voltage":0,"charger_pilot_current":32,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Blocking",t_request":32,"charge_current_request_max":32,"charge_port_led_color":"Off","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":null,"motorized_charge_port":false,"eu_vehicle":true,"timestamp":1485457690477}}: nosniff

The error can found at: "Blocking",t_request":32
This should be: "Blocking","charge_current_request":32

Also note the end of the data: ":no sniff" is data which shouldn't be there.

I think both problems are related.

delcomp, 2017-01-27:
This morning I changed SOCKET_BUFFER_MTU (1446u) to (1500u) in socket_buffer.h.
Now the problem seems to be solved, running 15 minutes without issues now. I will continue to do tests today.

I've adapted the test code to make it more readable:
        k = MAX_RESPONSE-1;
        i = 0;
        while (https.connected()) {
          j = https.read((uint8_t *)&sResponse[i], k);
          if (j > 0) {
            k -= j;
            i += j;
            sResponse[i] = '\0';
            Serial.print("Response: ");
            Serial.println(sResponse);
          }
        }

With an MTU of 1446 I get the following print's:
Bote the error once more:
- the first read results in a string ending with "Blocking",
- the second read appends data to the string: "Blocking",t_request"
This is wrong it should be: "Blocking","charge_current_request"
Note that a number of chars are 'lost' and some extra nonsense characters are appended at the end of the read string.
Setting the MTU to 1500 solves this issue, but I do not understand why. There must be some buffer overflow somewhere.

Response: {"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":null,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"<invalid>","battery_range":90.22,"est_battery_range":54.35,"ideal_battery_range":72.18,"battery_level":31,"usable_battery_level":30,"battery_current":0.0,"charge_energy_added":35.54,"charge_miles_added_rated":138.0,"charge_miles_added_ideal":110.5,"charger_voltage":0,"charger_pilot_current":32,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Blocking",

Response: {"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":null,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"<invalid>","battery_range":90.22,"est_battery_range":54.35,"ideal_battery_range":72.18,"battery_level":31,"usable_battery_level":30,"battery_current":0.0,"charge_energy_added":35.54,"charge_miles_added_rated":138.0,"charge_miles_added_ideal":110.5,"charger_voltage":0,"charger_pilot_current":32,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Blocking",t_request":32,"charge_current_request_max":32,"charge_port_led_color":"Off","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":null,"motorized_charge_port":false,"eu_vehicle":true,"timestamp":1485501915564}}: nosniff
X-T

delcomp, 2017-01-27:
Forgot one more thing.
With an MTU of 1446 I have this problem, but it's not always. Out of 3 reads, 1 read fails, the other 2 are perfectly ok.

sandeepmistry, 2017-02-16:
Hi @delcomp,

Thanks for reporting this, could you please provide the following to help look into the problem:

1) A minimal sample sketch to reproduce the issue
2) If possible, some way for us to use the same test server

Then we can try to reproduce the issue locally.

delcomp, 2017-03-09:
Sorry, the sketch I use is too long. It'll take for me to reduce it as a sample. The same test server can not be used because it used my login/password and I do not want to put those overhere.

But you can very easily test it. Just reduce the MTU to 50 bytes or so. Then access a random server and try to read more then 50 bytes. You'll notice some data in between 50 byte blocks is skipped. It's with all servers I've tested.

But again, changing the MTU to 1500u does solve the problem.

sandeepmistry, 2017-04-06:
> But you can very easily test it. Just reduce the MTU to 50 bytes or so. Then access a random server and try to read more then 50 bytes. You'll notice some data in between 50 byte blocks is skipped. It's with all servers I've tested.

This is not a good idea, if `SOCKET_BUFFER_MTU` is reduced below `1446` you'll also need to increase `SOCKET_BUFFER_TCP_SIZE` to a multiple of `SOCKET_BUFFER_MTU`. For example on AVR boards we have:

```
#define SOCKET_BUFFER_MTU						(16u)
#define SOCKET_BUFFER_UDP_SIZE					(SOCKET_BUFFER_UDP_HEADER_SIZE + 64u)
#define SOCKET_BUFFER_TCP_SIZE					(64u)
```

Again, a sketch to reproduce the original issue is very important.

sandeepmistry, 2017-05-24:
@delcomp ping, any updates on this? Do you have a sketch that can be used to reproduce this?

sandeepmistry, 2017-10-18:
Closing for now due to lack of feedback.

closed 2017-10-18 by sandeepmistry