arduino-libraries/WiFi101 #139: Library issues with Atmel Studio 7

broddo, 2017-01-27:
Hi, I've been using the WiFi101 library 0.12.1 with the Adafruit ATWINC1500 WiFi Breakout for some time now. I've used it with both the Arduino IDE (v 1.8.1) and the Sloeber plugin for Eclipse. 

Recently, however, I've had to debug my code in more detail so I started using Atmel Studio 7 along with the Atmel ICE debugger. 

The library compiles without problems in AS7 but it doesn't function correctly. 
For example, it will not connect to any network - it just times out in this loop in WiFiClass::startConnect()

`	// Wait for connection or timeout:
	unsigned long start = millis();
	while (!(_status & WL_CONNECTED) &&
			!(_status & WL_DISCONNECTED) &&
			millis() - start < 60000) {
		m2m_wifi_handle_events(NULL);
	}
`

In AP mode, it does actually manage to create the AP and I can connect to it - but the status never changes from WL_AP_LISTENING to WL_AP_CONNECTED. Also, calling WiFi.RSSI() always returns 0 (again due to a timeout). 

Can you think of any reason why this would be the case in AS7 versus the Arduino IDE? I'm guessing it could be due to compiler versions, configuration settings, optimizations etc.

broddo, 2017-02-10:
Ok, I've found the problem - well kind of. It appears that AS7 wasn't respecting the (weak) attributes attached to the functions in mm_bsp_arduino.c. Specifically these ones:
```
void __attribute__((weak)) attachInterruptMultiArch(uint32_t pin, void *chip_isr, uint32_t mode)
```
and
```
void __attriubte__((weak)) detachInterruptMultiArch(uint32_t pin)
```

So these weak functions in mm_bsp_arduino.c were being called instead of the strong equivalents in mm_bsp_arduino_avr.c. This occurs despite having the symbol -DARDUINO_ARCH_AVR as an argument to gcc and g++.

However, when I commented out the weak functions so that there would be no choice but for the strong functions to be resolved - it still wouldn't work and the problem appeared to be that the interrupt wasn't being set properly

For reference, I'm running this on an Arduino Mega2560. 

A bit of debugging in attachInteruptMultiArch() function and I found the problem. 

attachInterruptMultiArch() takes a uint32_t as its first parameter for the pin. From this point on, the pin is casted to an int. This would be relatively harmless in a 32-bit environment but the Mega2560 is 8-bit and an int on the Mega is only 16-bit. This cuts off the first two bytes of the pin variable (which contains the pin information as the Mega is little endian). So the rest of the function is dealing with an arbitrary number which will fail all the tests resulting in the interrupt not being set. 

The only option for the code here is to use the attachFakeInterruptToTimer() function should actually should be unnecessary in this situation. 

So the quick fix would be change the uint32_t of the pin argument to an int instead - assuming that doesn't screw up other architectures.

However, I had to go one step further, because the Mega needs the pinMode set as well. My quick fix was simply:
'''
void attachInterruptMultiArch(int pin, void *chip_isr, uint32_t mode) 
{
  pinMode(pin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(pin), chip_isr, mode);
}
'''
I removed the rest of the code as I don't need it in my case - I only need this code to work on the Mega. 

If I get time, I'll write a proper fix that supports the rest of the architectures and issue a pull request.

ericbaril72, 2017-07-17:
Hello Brodo, hopefully you receive this ( even months after your initial posting ).

I too work with Atmel Studio 7 and trying to get my hands around the library.
maybe you would have a suggestion for me.  I also use the Adafruit ATWINC1500 ( but the adafruit Feather M0 ).

Question--> **How to pause/resume the WINC1500 interrupts ?**
WINC1500 used as a client and wifi.status()==WL_CONNECTED

I am trying to make make it "non-blocking" AND share the SPI port.
I have an timer interupt (tc3 ) every 10ms to read a different device on the same SPIbus.

Hypothesis:
          Is the WINC1500 using non-maskable interupts ?
          Is the WINC1500 using a DMA transfer that has higher-priority then the SPI-beginTransaction ?

my timer interrupt check that the WINC1500_CS is not asserted and calls:
     SPI.beginTransaction(settings);               // should disable interupts
         Second_SPI_device_CS assert
         few SPI transactions
     SPI.endTransactions();                           // WINC1500 should resume where it was.

it works most of the time ...  I see using an oscilloscope that the WINC1500_CS are postponed.
I also see that I NEVER have the 2 Chip-Selects happening at the same time on the SPI bus.

BUT once in a while,my application freezes.   Which never happens if I do not let the secondary device access the SPI bus during any of the WINC1500 calls ( using a global flag ).

the WINC1500 access is in the main loop() and very simple.   Calls for an TCP/IP connection every 3 seconds:

if(client.connect()){
    client.print(postHeader);
    delay(100);
    int connectionTimeout=5000;
    while (connectionTimeout >0 ) {
        while(client.available())  char c = client.read();
        delay(100);
        connectionTimeout -=100;
        if (!client.connected()) break;
    }

}
client.stop();
     
------------------
In case your are asking:
postHeader specificly calls for the server NOT to keep connection open 

sprintf(postHeader, "POST /datapost HTTP/1.0\r\n"
						"Host: %s:%d\r\n"
						"Content-Type: application/json\r\n"
						"Content-Length: %d\r\n"
						"Connection: close\r\n\r\n"
						, URL.host, URL.port, jsonlength);

broddo, 2017-07-18:
Hi ericB - I may not be able to help you very much, as I only dug as deep into the WINC1500 driver code as I needed to solve my own problem with the Atmega 2560. I haven't tried to use the wifi module with an ARM part as of yet which means I ignored a good chunk of the low level hardware configuration as it didn't apply. Its a lot easier to configure an AVR than an ARM! 

> Is the WINC1500 using non-maskable interupts ?
Certainly not in my case - the Atmega 2560 doesn't have an NMIs that I'm aware of (except maybe reset and watchdog?). But a quick search through the library does reveal some references to NMIs

> Is the WINC1500 using a DMA transfer that has higher-priority then the SPI-beginTransaction ?
Again, I'm not sure! The Atmega 2560 isn't capable of DMA and so I didn't go looking for it.  

Try probing the IRQ pin as well - it would be interesting to see what role it has to play. 

A couple of questions:
Do you have a debugger? 
Can you share your interrupt routine?
When you say your application freezes, is it triggering a trap? Or is it just caught in an infinite loop somewhere?

ericbaril72, 2017-07-18:
Hello Brodo, Thanks for you quick answer.  
Yes, I did monitor both interrupts lines and both CS lines ( winc1500 and secondary device ).

From what I saw, the WINC1500 drivers do not instantly execute upon the IRQ line... which tells me there is probably a FLAG being set or a Separate "Timer" IRQ that polls the IRQ line.

I managed to use FLAGS to block my secondary SPI device for
client.available(), client.read(), client.connected(), client.stop()   as theese do not block the SPI for a long time.  I'll focus my attention on client.connect() which can block for quite a while.

Thanks for the suggestion about the debugger,  I've mainly used it for programming and had some trouble long long time ago with interupts during debugging so stopped using it as my code ALWAYS run on the first compile   ;-)

---
from my readings on the WINC1500, I saw it uses SPI.beginTransaction()/.endTransaction for each "data transaction".  So I thought that by using it, it blocks the other interrupts ( including the WINC1500 IRQs ).

my interrupt routine:
-- 
-- if WINC1500_CS == HIGH -> No WINC1500 transaction is happening
                             ( winc1500 drivers does :   Set CS LOW, stop interrupts via SPI.beginT... )
    {
    Set device_CS to LOW
    SPI.beginTransaction()
     SPI.data TX/RX
    SPI.endTransaction()
    }

So my problem is not ARM related as it is WINC1500 drivers oriented.  Since 99% of the users seems OK with the normal "blocking-mode", not that many people go into depth of the WINC1500.

And the Arduino Wrapper add a level of complexity over the ASF supplied WINC1500 code.

seeing that Googl'ing for   "attachinterrupts chip_irs" got me here ....  I saw you did go deeper then the 99%.

I hope I wont have to dig into ARM interrupts priorities ( device_INT being interrupted by WINC1500 )
as the crowd using this gets even smaller.

Hopefully I find soon. 
I always assume in my applications that Internet connection is flaky.  So I can't afford my stuff to be SO internet dependant ( as stopping for a few seconds while the server does not answer right away ).

ericbaril72, 2017-07-18:
Found it :           nm_bsp_interrupt_ctrl(0   or   1)
and also found that:
 hif_handle_isr()  polls the WINC continuously after an interrupt   (m2m_hif.c )

so I decided to cheat a little ... and modified the WINC1500 drivers. 
Added a global FLAG into the "connect" fonction.
the **client** does not hold a STATE per say ... I could possibly modify it to have a "PENDING_CONNECTION" state.   This could lead to good ASYNC use of the WINC100.

Processor never froze again .... Woohoo

**HOW-TO:**
(global flag)
int AwaitingConnectionFlag;

**-->in file  WINC1500Client.cpp<--   function     int connect()**
AwaitingConnectionFlag=true;                                      // Add this line
while (!IS_CONNECTED && millis() - start < 3000) {    // I changed my timeout to 3 seconds
		m2m_wifi_handle_events(NULL);
}

**--> file m2m_hif.c <--          function sint hif_handle_isr(void)**

while (gu8Interupt){
      if (AwaitingConnectionFlag) AwaitingConnectionFlag=false;         // added this line
      gu8Interrupt--;

ScottMackay2, 2017-08-18:
I am using the: 
Wifi board: ATWINC1500-MR210PB
MCU: Atmega 2560
Tool: Atmel Studio

I did the steps that broddo did to fix the library for the Atmega 2560.

But the problem I have now is:
My connected irq pin is not in the list of the digitalPinToInterrupt(). This makes me have to do the "strategy 1" step in the "attachInterruptToChangePin()" function. My irq is connected to digital pin 13, and that seems to be in the list of "digitalPinToPCICR()". So "strategy 1" should work after changing the uint32_t to int.

Sadly, after the attachInterruptMultiArch() has executed the program seems to crash (restarts the whole arduino and does it again) in the following part:
file: WiFi.cpp
function: WiFiClass::init()
line: m2m_periph_gpio_set_val(M2M_PERIPH_GPIO16, 1); // I also note that m2m_periph_gpio_set_val(M2M_PERIPH_GPIO15, 1); is called before that and that doesn't go wrong somehow.

So from there I went further down the call stack until I found:
file: nmspi.c
function: spi_write_reg(..., ...)

and somewhere there it crashed. I don't have a debugger so I am literally just pasting in lines to turn on leds to check if the code gets places. And the strange thing I noticed in the  spi_write_reg function, is that it is called once and succeeds till the end, then it is called a second time, where the code stream I followed has no single indication of that it is called again. (interrupt from another thread or something. Something to do with the irq?). Then it hangs and crashes.

I can go further but maybe someone knows what is going on before I waste another 10 hours because I am an embedded noob...

broddo, 2017-08-31:
@ScottMackay2 I don't meant to give you a short answer, but there could be any number of reasons why your code is crashing and trying to find it could be like trying to find a needle in a haystack. You'd be a lot faster by picking up a Atmel-ICE debugger, which isn't too expensive, and stepping through your code. If you're going to be doing a lot of development using Atmel products, its worth the investment!

broddo, 2017-08-31:
A quick update to this. Today I started using the latest version of the library (0.14.3). Before porting my few changes into the library, I thought I'd give it a test - and low and behold, it attached the interrupt to the pin just fine! (i.e. it wasn't resorting to the fake interrupt).

I can't see any specific changes that would have resulted in this fix - but also, when reviewing my initial comment, I released that it doesn't make sense anyway. Casting a 32-bit value to a 16-bit value should preserve the LSBs so that line should be harmless.

closed 2017-08-31 by broddo