arduino-libraries/WiFi101 #140: using gettime() with a static IP address

providersolutions, 2017-01-27:
any reason we can't use a static address and gettime() together?

gettime() always returns a zero if a WiFi.config(ip) is called prior to the WiFi.begin()

if i comment out the .config, gettime() works fine.

sandeepmistry, 2017-01-27:
@providersolutions I believe you'll also need to provide the IP address of a DNS server:

```
WiFi.config(local_ip, dns_server);
```

Otherwise the WINC1500 firmware will not be able to resolve the NTP servers IP address via DNS.

Can you please try this?

providersolutions, 2017-01-27:
Works!  thanks, been trying to figure it out for most of the day...thanks for the quick reply!

sandeepmistry, 2017-01-27:
Your welcome, glad I could help!

I'm going to close this issue now. @akash73 what do you think about updating the WiFi101 library reference to mention `WiFi.getTime()` requires a DNS server set?

closed 2017-01-27 by sandeepmistry

providersolutions, 2017-01-27:
i think that would be very helpful to others.

mamama1, 2017-01-28:
you gonna explain the internet now in the reference?

providersolutions, 2017-01-28:
good documentation should explain dependencies and assumptions.   it is not obvious why, with a static address specified in “x.x.x.x” format, begin() works without a DNS specified in config() but gettime() does not.

gettime():

when using config() to specify a static ip address, gettime() requires that config() also specify a DNS address.

   
     
that gonna work for ya?