arduino-libraries/WiFi101 #141: Get all BSSID around and their channels

gmag11, 2017-01-28:
Current library version does not implement a method to get BSSIDs and channels after a scan is called, only SSIDs and RSSIs are allowed.

I've implemented them and will make a pull request.

sandeepmistry, 2017-02-16:
Hi @gmag11,

I've submitted #143 for this, it's similar to #142 which you submitted, however it has:

* a change to the `BSSIS` API to pass a buffer in like `macAddress` and `BSSID(bssid)`.
* the styling is more consistent with the rest of the code in the WiFi101 library.

Could you please try it out and report your feedback?

gmag11, 2017-02-18:
Hi @sandeepmistry 

I think a function the form as `uint8_t* BSSID(uint8_t pos)` is consistent with already existing `char* SSID(uint8_t pos)`. 
What is pointer parameter in `uint8_t* BSSID(uint8_t pos, uint8_t* bssid)` intended for?

sandeepmistry, 2017-02-21:
> What is pointer parameter in uint8_t* BSSID(uint8_t pos, uint8_t* bssid) intended for?

It's for the 6 byte array the caller wants to store the BSSID in, please see:
 * https://www.arduino.cc/en/Reference/WiFi101BSSID
 * https://www.arduino.cc/en/Reference/WiFi101MACAddress

> I think a function the form as uint8_t* BSSID(uint8_t pos) is consistent with already existing char* SSID(uint8_t pos).

Right, but `char* SSID(uint8_t pos)` returns a C string instead of a `uint8_t*`.

RedDragon9, 2017-02-27:
I would also like to be able to get the channel of the Wifi networks after a scan.

gmag11, 2017-02-27:
With my code you can get bssid and channel of all surrounding networks, besides ssid and rssi that were already possible. Check my fork and share your results, thanx.

RedDragon9, 2017-02-27:
Could you post the link to the code please.

gmag11, 2017-02-27:
https://github.com/gmag11/WiFi101 Select scan_info branch

RedDragon9, 2017-02-28:
I added the library but can you tell me the code I need to add to the scan networks program to show the channal and bssid because I can't seem to get it to work or do you have a scan network explain that includes that?

gmag11, 2017-02-28:
It's used in this code https://github.com/gmag11/ESPWifiLocation/blob/master/src/WifiLocation.cpp#L19

RedDragon9, 2017-02-28:
Ok thanks, I got it working. I found I still had the wifi101 library installed and that why the code wouldn't work.

agdl, 2017-03-01:
closed via https://github.com/arduino-libraries/WiFi101/pull/146

closed 2017-03-01 by agdl