arduino-libraries/WiFi101 #144: After a certificate_request, if no certificates are available, clients now MUST send an empty certificate list (RFC 5246)

jorupp, 2017-02-20:
In trying to use the Adafruit M0 with Azure IoT over MQTT-SSL, but having issues opening the connection.  @Eskibear dug a bit deeper than I did and found that in this one case, the server was requesting certs, but [the client wasn't providing any](https://github.com/Azure/azure-iot-arduino-protocol-mqtt/issues/3#issuecomment-274389293) (not even an empty list), which seemed suspicious.

Is this a firmware issue or a driver issue?

sandeepmistry, 2017-02-21:
Hi @jorupp, @Eskibear,

This sounds like a firmware issue to me. Could you please open a support case with Atmel Support then update this issue with the case number so it can be followed up by us later if needed. Thanks.

http://www.atmel.com/design-support/

sandeepmistry, 2017-02-21:
Sorry, accidentally closed this, re-opening ...

jorupp, 2017-02-21:
We'll see if I filled out that support form right - got a case number from their Salesforce site - 00051323.

sandeepmistry, 2017-02-21:
@jorupp excellent, thanks for letting us know!

sandeepmistry, 2017-03-13:
@jorupp any news from Atmel/Microchip on this?

There's a PR build of the IDE that contains the new v19.5.2 WINC1500 firmware: https://github.com/arduino/Arduino/pull/6069#issuecomment-286223342. Any one up for testing it with the master version of the library to see if this has been fixed?

Eskibear, 2017-03-16:
@sandeepmistry Thanks, this new firmware works for Azure-MQTT. 
I can establish the SSL connection with AzureIoTHub using port 8883 usiing Feather M0 now.

sandeepmistry, 2017-03-16:
@Eskibear @jorupp great to hear! Thanks for trying it out!

I'm closing this as resolved.

closed 2017-03-16 by sandeepmistry