arduino-libraries/WiFi101 #145: WiFiSSL client works without downloading the certificate

sudheera8, 2017-02-21:
Hi

I have been using ATWINC1500 module with arduino zero to connect to IBM bluemix server using the Pubsubclient library for the  MQTT stack. I downloaded the certificate corresponding to the server and used the WiFiSSLClient to connect to the encrypted port(8883) for TLS communication.  I then took another ATWINC1500  module used the SSL client to connect using the 8883 port it did establish the connection successfully and sent the payloads even without downloading the certificate.

How the communication being encrypted without downloading the certificate? Is there any way I could verify? This is critical for the completion of my project. Kindly respond

Thanks
Sudheera

sandeepmistry, 2017-02-21:
Hi @sudheera8,

The firmware shipped with the MKR1000 and WiFi 101 shield comes preflashed with the following root certificates: https://github.com/arduino-libraries/WiFi101-FirmwareUpdater/tree/master/certs

https://mybluemix.net uses the DigiCert root certificate, which is included in the link above.

[image]

closed 2017-02-21 by sandeepmistry

sudheera8, 2017-02-21:
Hi sandeepmistry 

I appreciate your quick response. If this is the case, then is it fair to tell my clients that their data is secure during the communication? Else Is there anyway I could monitor the packets going out of the device before I put it in the market?

Thanks
Sudheera

sandeepmistry, 2017-02-21:
@sudheera8 you can setup an access point on your PC and then use something like Wireshark to confirm the data is encrypted.

sudheera8, 2017-02-21:
That makes sense. I will try to do that and get back to you.

Thanks again! :)