arduino-libraries/WiFi101 #149: changing the PA value

sudheera8, 2017-03-09:
Hi 

We are working on changing the TX power level from 18dNm to 12 dBm. I found these values:
typedef enum {
       TX_PWR_HIGH = ((uint8) 1),
       /*!< PPA Gain 6dbm   PA Gain 18dbm */
       TX_PWR_MED,
       /*!< PPA Gain 6dbm   PA Gain 12dbm */
       TX_PWR_LOW,
       /*!< PPA Gain 6dbm   PA Gain 6dbm */
}tenuM2mTxPwrLevel;

which are used in m2m_wifi_set_tx_power() function. But where are they set?  I need it to be TX_PWR_MED.

Kindly respond. I need it to be changed to pass the FCC for our wearable.

Thanks
Sudheera

chris0sbg, 2017-05-16:
Hi,

I have the same problem.
Do you found a solution for this problem?

Thanks
chris

chris0sbg, 2017-05-18:
Hi,

for my case I set
m2m_wifi_set_tx_power()
and
m2m_wifi_set_gains()
just after
m2m_wifi_init()
and it seems working.

I also asked Atmel for the difference of these two functions.
This was the answer:
The m2m_wifi_set_tx_power() is used to set the PA gain to 3 different levels while m2m_wifi_set_gains() is used to set the PPA gain before the transmission. These values along with the digital gain values will define module TX power for different channels and different data rate in 802.11 b/g/n modes. 

Maybe it helps someone...

Best regards,
Chris

sandeepmistry, 2017-05-23:
Please see @chris0sbg's answer above. At this time the WiFi101 library does not support changing the PA value. You'll need to use the lower level WINC1500 ASF API's directly.

closed 2017-05-23 by sandeepmistry

sudheera8, 2017-05-24:
Hi @chris0sbg 

typedef enum {
TX_PWR_HIGH = ((uint8) 1),
/*!< PPA Gain 6dbm PA Gain 18dbm /
TX_PWR_MED,
/!< PPA Gain 6dbm PA Gain 12dbm /
TX_PWR_LOW,
/!< PPA Gain 6dbm PA Gain 6dbm */
}tenuM2mTxPwrLevel;

I knew that tenuM2mTxPwrLevel was being used at the TX_PWR_HIGH level and hence I need to replace that usage with  the MED level.

In the above snippet I interchanged the TX_PWR_HIGH and TX_PWR_MED so that when ever tenuM2mTxPwrLevel is used, power lever would be set to medium. 

I apologize for replying late and I appreciate your insights. :) 

Thanks
Sudheera

sudheera8, 2017-05-28:
Hi @chris0sbg, @sandeepmistry 

Where is the m2m_wifi_set_gains() API declared and defined in the wifi101 library? Kindly let me know where I can find it. It would be really helpful. When I compiled thecode I get errors that it has not been declared.

Thanks
Sudheera

chris0sbg, 2017-05-28:
Hi Sudheera,

see:
master/src/driver/source/m2m_wifi.c
Line: 1006

Best regards,
Chris

sudheera8, 2017-05-28:
Hi  Chris

Thanks a ton! My version of the library isn't having that unction.Will update it. :)

When you mentioned digital gain were you talking about the gain table? Apparently there is a gain table in the WINC's flash and that is what I need to modify. I would appreciate any insights. 

Thanks
Sudheera

sudheera8, 2017-05-28:
Hi @sandeepmistry

 I see there is this API  m2m_ssl_send_certs_to_winc(); and I understand that it can be used to upload certs to the WINC module.  The parameters are the pointer to the certificate and and its size. Can I use this to send a certificate stored in controller's flash? 

I apologize for having so many questions, but there is no better way that I could complete these tasks in such a short span.

Thanks
sudheera