arduino-libraries/WiFi101 #150: Simultaneous WiFiServer and WiFiClient lockup MKR1000 due to using wrong sockets

delcomp, 2017-03-09:
My app makes continues connections to multiple servers, one-after-the-other, using the same WiFiClient object. Simultaneously a WiFiServer object is used to allow a webbrowser to read back data from the app. The webbrowser reads every second a status. A second WiFiClient object is used for this.
This should use two different sockets but it seems that is not always the case. Sometimes both WiFiClient objects get the same socket assigned which eventually locks-up the program.

WiFiServer = socket 0 (assigned at start of program)
WiFiClient = socket 1 (this socket connects/stop continuously, I mean it's not al the time assigned to WiFiClient)
Second WiFiClient from WiFIServer object = socket 1, that happens even when socket 1 has already been assigned !!

I spend the whole day trying to find the reason for this, but it seems to be related to something in WINC1500. I found a (not so nice) solution, so I'll stop searching. But this should be solved in the library.

My code is much too long to paste overhere, but I'll explain a bit more in detail what happens and how I solved it.

WiFiServer::available will return a WiFiClient object when the SOCKET_BUFFER_FLAG_SPAWN flag is set. This is done by socketBufferCb (in socket_buffer.c) in case SOCKET_MSG_ACCEPT. This callback-function also sets a socket number which (I'm not sure about this) seems to be returned to the callback, probably coming from the WINC1500 firmware (?).
Anyhow the socket number returned is sometimes the same socket as the one already assigned to the WiFiClient object.

I solved this by patching socket.c, the socket(uin16 u16Domaind, ....) function.
This function is used to request the next free socket and it starts searching from socket 0. I changed this so it starts searching from socket 1 (u8Count = 1 iso 0).

The result:
WiFiServer object = socket 1 (fixed, and assigned at start of the program)
WiFiClient object = socket 2 (this socket connects/stop continuously during the program run)
Socket 0 = always returned by the socketBufferCb/SOCKET_MSG_ACCETP callback, so the server WIFIClient socket is always 0.
No more overwrites of the sockets, program works fine.

delcomp, 2017-03-09:
Above solution does not hold with multiple simultaneous webserver request. So I've searched further and have now found the underlying issue.

Each connect reserves a socket in the Atmel code (from 0 to 6). But this stack of sockets is not synced with the similar WINC1500 socket stack. When a browser sends a request to the WINC1500 it will reserve the first available socket and report this with the callback function to the Atmel. But there is a delay in doing this. During this delay the Atmel might also have reserved the same socket in response to a connect. Result is that one socket is reserved for two purposes and the system eventually crashes.

The easiest solution is to change the SOCKET socket(uint16 u16Domain, uint8 u8Type, uint8 u8Flags) function in socket.c. Just reserver sockets from the socket stack from TCP_SOCK_MAX-1 downwards, in stead of from 0 till TCP_SOCK_MAX-1. 
This way the Atmel will always reserve sockets 6, 5, ... and the WINC1500 will reserve sockets 0, 1, ....
Reserving the same socket on both sides becomes rare with this setup, although not impossible. To make things even better the last available socket should never be reserved by the Atmel. But that's only needed in a setup with several clients (my app uses 2 simultaneously) and several webbrowsers accessing the server function (in my app appr. 3 will be worst case).

delcomp, 2017-03-09:
Changed code in socket.c 
(socket 0 can not be used, but for my app that's no problem)

		if(u8Type == SOCK_STREAM)
		{
			u8SocketCount = 0;
			u8Count = TCP_SOCK_MAX-1;
		}
		else if(u8Type == SOCK_DGRAM)
		{
			/*--- UDP SOCKET ---*/
			u8SocketCount = TCP_SOCK_MAX-1;
			u8Count = MAX_SOCKET-1;
		}
		else
			return sock;

		for(;u8Count > u8SocketCount; u8Count --)
		{

sandeepmistry, 2017-03-13:
@delcomp could you please share a simple sketch that we can use to reproduce the issue locally?

What version of the WiFi101 library are you using? Could this be related to https://github.com/arduino-libraries/WiFi101/issues/127 which was solved with #128? This was released in the v0.12.0 version from this January.

delcomp, 2017-03-21:
Hi I'm using the latest version 0.13.0. Also updated the firmware of the WINC1500 to the newest one included with 1.8.2 Hourly.

It's not easy to build a simple sketch to test this. If I find some time I'll give it a try.

sandeepmistry, 2017-05-24:
@delcomp ping any updates on this? Does it still happen with the latest WiFi101 release? Did you get a chance to make a sketch to reproduce the issue?

sandeepmistry, 2017-10-18:
Closing for now due to lack of feedback.

closed 2017-10-18 by sandeepmistry