arduino-libraries/WiFi101 #151: Example of receive post data

jdneo, 2017-03-13:
Hi,

I have a WINC1500 which acts as a server and need to receive the form post data from other client. After a long time searching, I failed to find the example of this senario.

Currently, if I use WiFiClient::read() method, I can only receive the request header part. The form-data part is missing. Is there any way to receive the post data?

Thank you!

sandeepmistry, 2017-03-13:
@jdneo please share the sketch you are using.

jdneo, 2017-03-16:
Hi sandeepmistry,

Sorry for my late reply. The sketch is listed as follows:

```
#include <SPI.h>
#include <WiFi101.h>

int led = LED_BUILTIN;
const String WIFI_CONFIG_HEAD = "GET /setting?";

char ssid[] = "wifi101-network"; // created AP name
char pass[] = "1234567890";      // AP password (needed only for WEP, must be exactly 10 or 26 characters in length)
int keyIndex = 0;                // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
WiFiServer server(80);

void setup()
{
  Serial.begin(9600);
  while (!Serial)
  {
    ; 
  }

  Serial.println("Access Point Web Server");

  pinMode(led, OUTPUT); // set the LED pin mode

  if (WiFi.status() == WL_NO_SHIELD)
  {
    Serial.println("WiFi shield not present");
    while (true)
      ;
  }

  Serial.print("Creating access point named: ");
  Serial.println(ssid);

  status = WiFi.beginAP(ssid);
  if (status != WL_AP_LISTENING)
  {
    Serial.println("Creating access point failed");
    while (true)
      ;
  }

  delay(10000);

  server.begin();
}

void loop()
{
  if (status != WiFi.status())
  {
    status = WiFi.status();
  }

  WiFiClient client = server.available(); 

  if (client)
  {  
    Serial.println("new client"); 
    String currentLine = "";    
    while (client.connected())
    { 
      if (client.available())
      {                         
        char c = client.read();
        Serial.write(c);
        if (c == '\n')
        { 
          if (currentLine.length() == 0)
          {
           
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            client.println("<!DOCTYPE HTML>");
            client.println("<html>");
            client.println("<p>Hello from Feather M0!<P/><br/>");

            client.println("<form method='get' action='setting'>");

            client.println("<label>SSID:</label>");
            client.println("<input name='ssid' length=64 type='text'/><br/><br/>");

            client.println("<label>PASS:</label>");
            client.println("<input name='pass' length=64 type='password'/><br/><br/>");

            client.println("<input type='submit'/>");

            client.println("</form>");
            client.println("</html>");

            client.println();
            break;
          }
          else
          {
            if (currentLine.startsWith(WIFI_CONFIG_HEAD))
            {
              /* parse the get request data here*/
              break;
            }
            currentLine = "";
          }
        }
        else if (c != '\r')
        {                   
          currentLine += c; 
        }
      }
    }
    client.stop();
    Serial.println("client disconnected");
  }
}
```

It's very similar with the [sample code](https://github.com/arduino-libraries/WiFi101/blob/master/examples/AP_SimpleWebServer/AP_SimpleWebServer.ino). Currently the form using get action and I can successfully parse the request. But when I change the action field to _POST_

`client.read();`

can only get the request header part. The body part(form data) is missing. Is there any way to parse the post request?

Thanks

sandeepmistry, 2017-03-16:
@jdneo here's a modified version:

```arduino
#include <SPI.h>
#include <WiFi101.h>

int led = LED_BUILTIN;
const String WIFI_CONFIG_HEAD = "GET /setting?";

char ssid[] = "ssid"; // ssid
char pass[] = "pass";      // password
int keyIndex = 0;                // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
WiFiServer server(80);

void setup()
{
  Serial.begin(9600);
  while (!Serial)
  {
    ; 
  }

  Serial.println("POST Web Server");

  pinMode(led, OUTPUT); // set the LED pin mode

  if (WiFi.status() == WL_NO_SHIELD)
  {
    Serial.println("WiFi shield not present");
    while (true)
      ;
  }

  Serial.print("Connecting to access point named: ");
  Serial.println(ssid);

  status = WiFi.begin(ssid, pass);
  if (status != WL_CONNECTED)
  {
    Serial.println("Connecting to AP failed");
    while (true)
      ;
  }

  IPAddress ip = WiFi.localIP();
  Serial.print("Connected to AP, IP is: ");
  Serial.println(ip);

  delay(10000);

  server.begin();
}

void loop()
{
  if (status != WiFi.status())
  {
    status = WiFi.status();
  }

  WiFiClient client = server.available(); 

  if (client)
  {  
    Serial.println("new client"); 
    String currentLine = "";    
    while (client.connected())
    { 
      if (client.available())
      {                         
        char c = client.read();
        Serial.write(c);
        if (c == '\n')
        { 
          if (currentLine.length() == 0)
          {
// NEW ///////////////////////////////////////////
            String requestBody;

            while (client.available()) {
              requestBody += (char)client.read();
            }

            if (requestBody.length()) {
              Serial.println("Request body = ");
              Serial.println(requestBody);
              Serial.println();
            }
////////////////////////////////////////////////            
           
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            client.println("<!DOCTYPE HTML>");
            client.println("<html>");
            client.println("<p>Hello from Feather M0!<P/><br/>");

            client.println("<form method='post' action='setting'>");

            client.println("<label>SSID:</label>");
            client.println("<input name='ssid' length=64 type='text'/><br/><br/>");

            client.println("<label>PASS:</label>");
            client.println("<input name='pass' length=64 type='password'/><br/><br/>");

            client.println("<input type='submit'/>");

            client.println("</form>");
            client.println("</html>");

            client.println();
            break;
          }
          else
          {
            if (currentLine.startsWith(WIFI_CONFIG_HEAD))
            {
              /* parse the get request data here*/
              break;
            }
            currentLine = "";
          }
        }
        else if (c != '\r')
        {                   
          currentLine += c; 
        }
      }
    }
    client.stop();
    Serial.println("client disconnected");
  }
}
```

There's nothing wrong with the WiFi101 library, the request body is available after the headers are sent and the end of the headers are indicated by a blank like.

closed 2017-03-16 by sandeepmistry

jdneo, 2017-03-17:
Thank you @sandeepmistry. Seems I asked a stupid question  ;-)

DannyV1, 2017-04-08:
I found that I needed to waste the first hidden (NULL) byte:
(I'm using the String variable postData instead):
`
    ....
    String postData = "";
    //WASTE THIS BYTE: should be null
    char nByte = (char)client.read();
    if (nByte != '\0') {
        postData = nByte;
    }
    while (client.available()) {
        postData += (char)client.read();
    }
    if (postData.length()) {
      ....
`

Regards,
Danny

robertsonadams, 2017-04-22:
May I ask, what does the corresponding HTML form look like that takes in the posted data?

DannyV1, 2017-04-23:
Hi Robert,
Firstly I made a mistake on my previous post, the section labelled "WASTE THIS BYTE" is an error.
Here's an excerpt of my code, and a picture of my 'Whistle' button at the end: 
<pre><code>
//Process hidden POST data
void processPostData(String pstd,  int cLen)
{
const byte POSTBUFF_SIZE = 60;
char  postBuff[POSTBUFF_SIZE + 1];
const uint8_t EB_LEN = 180;
char ErrBuff[EB_LEN + 1];
char *pos = NULL;
char cmdState;
uint8_t pbLen = pstd.length();
uint8_t mainStrLen = strlen(MAIN_STR);

    memset(postBuff, 0, POSTBUFF_SIZE);
    for (int ii = 0; ii < cLen && ii < POSTBUFF_SIZE; ii++) {
        if (ii >= pbLen) {
            if (ii < cLen - 1) {
                snprintf(ErrBuff , EB_LEN,
                    "Not all POST data rxed [%s] (%d of %d)",
                                postBuff, ii, cLen);
                StatusbarStr = ErrBuff;
                return;
            }
            break;
        }
        postBuff[ii] = pstd[ii];
        postBuff[ii+1] = '\0';
    }

    if ((pos=strstr(postBuff, MAIN_STR)) != NULL)   {
        //MAIN_=H
        pos += (mainStrLen +1);
        if (pos != NULL) {
            cmdState = *pos; //"H"
            //give web page feedback that command has been processed
            StatusbarStr = "Status: ";
            StatusbarStr += cmdState;
            StatusbarStr += "  ";
            switch (cmdState) {
                case 'H':
                //Whistle
                    digitalWrite(LED_PIN, HIGH);
                    digitalWrite(WHISTLE_PIN, HIGH);
                    delay(100);
                    digitalWrite(WHISTLE_PIN, LOW);
                    delay(400);
                    digitalWrite(LED_PIN, LOW);
                    //remove web statusbar msg after 30 secs
                    //(handled in loop)
                     cmdRefreshed = millis();
                    break;
                case 'L':
                    break;
                case 'C':
                    break;
                case 'B':
                    break;
            } //switch
        } //if (pos != NULL) {
    } //pos=strstr(postBuff, MAIN_STR
}

// called from inside loop()
void checkForClient()
{
bool isPost = false;
bool finished = false;
[... 134 lines cut ...]
                }
                if (!finished  && currentLine.startsWith("POST")) {
#ifdef DEBUG2
                    Serial.println(F("POST "));
#endif //DEBUG2
                    isPost = true;
                }
            } //if (client.available(
            if ((count--) <= 0) {
                Serial.println("Timeout!");
                break;
            }
        } //while (client.connected()

        // close the connection:
        delay(2); // give the web browser time to receive the data
        client.stop();
#ifdef DEBUG1
        Serial.println("client disconnected");
#endif
    } //if (client
} //checkForClient
[image]

</pre></code>