arduino-libraries/WiFi101 #154: Switching between two WiFi networks does not work

delcomp, 2017-03-21:
Using the MKR1000 with latest libs and 1.8.1. I've also upgraded the firmware of the MKR1000 the latest (1.8.2 Hourly included).

My app should be able to switch between different WiFi networks dynamically. The device is moved around and should for example connect at work to the 'Work' network and at home to the 'Home' network. It's perfectly possible to detect not-connected and connected. But reconnecting to a new WiFi network seems to be unreliable. It sometimes works, most of the time fails. I've written a small test sketch. It just continuously swaps between two networks (you need to change the defines of course). In most case it will connect to one network and never to the other, which one it connects to is random.

SSID1 failed
SSID2 connected
1
1
SSID1 failed
SSID2 connected
1
1

#define SSID1 "SSID_1"
#define PW1 "PW_1"
#define SSID2 "SSID_2"
#define PW2 "PW_2"

#include <WiFi101.h>

WiFiClient client;
WiFiClient webclient;
WiFiServer server(80);

void setup() {
unsigned long l;

  Serial.begin(115200);                                       // Initialize serial and wait for port to open:
  l = millis() + 3000;
  while (!Serial && (l > millis()));                          // replaces: delay(3000);
}

void loop() {
int i;

  WiFi.begin(SSID1,PW1);
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("SSID1 connected");
    i = server.beginSSL();
    Serial.println(i);
    i = client.connectSSL("api.thingspeak.com", 443);
    Serial.println(i);
    client.stop();
    delay(5000);
    WiFi.end();
    WiFi.init();
  } else Serial.println("SSID1 failed");

  WiFi.begin(SSID2,PW2);
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("SSID2 connected");
    i = server.beginSSL();
    Serial.println(i);
    i = client.connectSSL("api.thingspeak.com", 443);
    Serial.println(i);
    client.stop();
    delay(5000);
    WiFi.end();
    WiFi.init();
  } else Serial.println("SSID2 failed");
   
}

I've done many tests on this. Adding the WiFi.end()/Wifi.init() sequence makes it a little bit more reliable, but it does still fail in 75% cases.

delcomp, 2017-03-21:
Some additional information.
Above problem occurs when I use my Apple Airport Express with two WiFi networks enabled:
1. The main network 'Apple'
2. The guest network 'Guest'
As mentioned swapping between those two does not work. Note that all my other equipment easily switches between the networks (iMac, PC, MacBook, iPhone, ...)

I now did another test using two other routers, each configured with a different SSID: 'ASUS' and 'iPhone'.
Swapping between 'ASUS', 'iPhone', 'Apple' works always. But the 'Guest' network is rejected.
If I swap between 'ASUS', 'iPhone', 'Guest' (without ever using 'Apple') is also works, but 'Apple' will be rejected.
By powering on/off the Airport Express the thing is reset. But again it will only connect to the first network on the AE.

sandeepmistry, 2017-04-06:
Hi @delcomp,

I'm not able to reproduce this locally, I used the sketch you provided but removed the `WiFi.init()` statements with an Apple Airport Express and iPhone as a hotspot:

```
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
SSID1 connected
1
1
SSID2 connected
1
1
```

Other details:
* WINC1500 f/w 19.5.2 on a MKR1000 with WiFi101 library v0.14.0

Do both the Guest network and Apple network have the same BSSID? Have you tried adding a delay after `WiFi.end()`?

sandeepmistry, 2017-05-10:
I'm closing this for now to to lack of feedback. Please re-open with requested info.

closed 2017-05-10 by sandeepmistry