arduino-libraries/WiFi101 #155: mDNS Responder Is Flawed

washcroft, 2017-03-29:
The handler code in WiFiMDNSResponder.cpp is flawed, it can't be working for most.

1) When comparing expectedRequestHeader, it doesn't allow for a varying DNS transaction ID (first two bytes - 0x00, 0x00)

2) Again, when comparing expectedRequestHeader it expects clients to always ask one DNS question per packet (bytes 3 and 4 - 0x00, 0x01), when in fact the A and AAAA request will often be asked at the same time.

3) When checking expectedRequestLength, it doesn't allow for additional data in the additional data section of the DNS packet (i.e. EDNS0 OPT)

sandeepmistry, 2017-04-07:
Hi @washcroft,

Is there a tool you can recommend we use to re-create these conditions?

We've tested with macOS and Linux, and have not observed these, but I agree they should be fixed.

Thanks.

sandeepmistry, 2017-04-07:
> When comparing expectedRequestHeader, it doesn't allow for a varying DNS transaction ID (first two bytes - 0x00, 0x00)

Not sure how accurate this site is: http://imjac.in/ta/post/2016/08/24/13-55-35-generated/ - but it states:

> The first 2 bytes are the Transaction ID. These bytes are always 0 in an mDNS packet. The next 2 bytes are the flags, for us, its 0x80 or 0x84.

sandeepmistry, 2017-04-07:
Hmm, the `systemd` source code has a similar statement: https://github.com/systemd/systemd/blob/master/src/resolve/resolved-dns-scope.c#L836

> /* For mDNS, the transaction ID should always be 0 */

washcroft, 2017-04-07:
OK I might have that piece wrong if that is indeed the case, however the other two appear to be valid.

A Windows 10 client (running the Apple bonjour client) request contained both A and AAAA questions.

A Raspbian client request contained the EDNS0 additional data section. Both instances alter what wold otherwise be expected in the packet.

sandeepmistry, 2017-04-07:
Thanks, I'll get a Win 10 setup with the Apple Bonjour client.

> A Raspbian client request contained the EDNS0 additional data section. Both instances alter what wold otherwise be expected in the packet.

Do you have some steps I can run do this? In the past I've just run the ping command: `ping wifi101.local`

sandeepmistry, 2017-04-07:
> Again, when comparing expectedRequestHeader it expects clients to always ask one DNS question per packet (bytes 3 and 4 - 0x00, 0x01), when in fact the A and AAAA request will often be asked at the same time.

#157 submitted as a proposal to address this.

washcroft, 2017-04-10:
Thanks, that fixes all the issues I was encountering with it.

washcroft, 2017-05-09:
Unfortunately, this appears to have introduced a new bug.

There is no upper limit checking on the received packet size, so the code can end up trying to assign a very large byte array for holding the packet, exhausting RAM and crashing.

I'm not sure what on my network is broadcasting such large packets, but it can happen intermittently.

https://github.com/arduino-libraries/WiFi101/blob/master/src/WiFiMDNSResponder.cpp#L139

Debugging this intermittent issue, often days inbetween crashes, I've seen packetLength be as high as 48000!

sandeepmistry, 2017-05-10:
@washcroft ok, I think we should add a check for a maximum packet length. Maybe make it smaller for AVR boards than other boards.

Any suggestions on what number we should use? I was thinking something between 2048 and 4096 ...

washcroft, 2017-05-11:
In this basic implementation of mDNS, we're only interested in the packet data up to minimumExpectedRequestLength, so that should be the size of the buffer, discard the rest?

sandeepmistry, 2017-05-12:
@washcroft thanks, that makes sense, I've implemented in PR https://github.com/arduino-libraries/WiFi101/pull/169.

sandeepmistry, 2017-06-01:
Closing this again, the new changes from PR #169 resolve them.

closed 2017-06-01 by sandeepmistry