arduino-libraries/WiFi101 #156: wifisslclient.available() returns false after downloading 40kB.

sudheera8, 2017-04-04:
Hi

I am trying to download a 75kB bin file using wifi101 and arduino M0 over http connection. The download is successful with wificlient(port 80) but with wifisslclient(port 443), the download often fails. The sslclient is connected but  sslclient.available() returns false after 40-50kB. Is there a limit on the size of the file that can be downloaded ?

I also tried a suggestion to increase the MTU size but that didn't solve my problem. Any help would be really appreciated.

Thanks
Sudheera

sandeepmistry, 2017-04-06:
Hi @sudheera8,

Do you have a minimal sketch that we can use to reproduce the issue?

`.available()` could be momentarily returning 0, have tried to call it again a bit later?

The [ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient) library might be of interest, it has some code to handle situations like this: https://github.com/arduino-libraries/ArduinoHttpClient/blob/master/src/HttpClient.cpp#L547-L586

sudheera8, 2017-04-09:
Hi @sandeepmistry 

Thanks for responding. I tried your suggestion. Using the Wificlient , .available() returns zero momentarily like you said.But with SSLclient it keeps returning zero continuously. 

However, after rigorous testing I found that the download is successful sometimes,we have also tried using a different server and it did work for larger files. So it might be the issue with the Bluemix server we have been using; so we have contacted the server hosts, I will let you know if I make any progress.

Thanks
Sudheera

sandeepmistry, 2017-04-10:
@sudheera8 thanks for the update. Should we close this issue for now? We can always re-open it in the future if needed.

sudheera8, 2017-04-10:
@sandeepmistry 

Sure. I will ask for a re-open if required.

Thanks For your quick responses. I really appreciate that.

sandeepmistry, 2017-04-10:
Ok, thanks!

I'll close this for now as we think it's a server side issue.

closed 2017-04-10 by sandeepmistry