arduino-libraries/WiFi101 #158: WiFi Release 0.14.0 Breaks UPD Multicast Sockets

youcangetme, 2017-04-12:
Hello!

My app is way too big and complex to post here but it relies on Multicast UDP reception in order to work. I just upgraded to the new WiFi101 Release 0.14.0 and the MKR1000 stop receiving packets (or it is receiving packets but never honoring the multicastUDP.parsePacket() call) . I degrade back to 0.13.0 and my app works just fine.

This feature (multicast sockets) is required for my app to "self configure" since it doesn't have a user interface during that stage of its lifecycle.

Thanks for looking into this for me!

If you require the app I can grant temporary access to the private GitHub project.

youcangetme, 2017-04-12:
For the record, I tried to flash the new firmware which succeeded BUT when I tried to roll it back to an older version it appeared to update fine but the byte check fails.

Anyways, new firmware didn't work at all. It appears to initialize and set up a network for the initial phase of the setup process. I can navigate to its web page but now the list of SSIDs doesn't render. So if I can't select a network to join I'm dead in the water which is why I rolled back the firmware update.

Anyways! I am using FlashStorage.h which was the cause of the problems last time. Not sure if this is a more expanded set of the same bugs?

At least I got it to roll back so I can work while ya'll figure out what went wrong.

sandeepmistry, 2017-04-18:
Hi @youcangetme,

I've opened PR https://github.com/arduino-libraries/WiFi101/pull/159 to attempt to fix it. However, it was a bit tricky for me to reproduce the issue consistently.

It would really speed things up if you could provide us with a minimal sketch and procedure that can be used to consistently reproduce this issue.

youcangetme, 2017-04-19:
This project is really complex and I wouldn't even know where to start marking a minimal sketch.

sandeepmistry, 2017-05-10:
@youcangetme have you tried the latest the version of the WiFi101 library?

> This project is really complex and I wouldn't even know where to start marking a minimal sketch.

Since the issue is related to MDNS, you could extract the MDNS specific code into a new sketch and provide it on this issue.

sandeepmistry, 2017-10-18:
Closing for now due to lack of feedback.

closed 2017-10-18 by sandeepmistry