arduino-libraries/WiFi101 #160: Bug in function WiFiServer::write

MarkusLange, 2017-04-28:
server.write does not work, function needs to change from

```
size_t WiFiServer::write(const uint8_t *buffer, size_t size)
{
	size_t n = 0;
	WiFiClient *client;

	for (int sock = 0; sock < TCP_SOCK_MAX; sock++) {
		client = WiFi._client[sock];
		if (client && client->_flag & SOCKET_BUFFER_FLAG_CONNECTED) {
			if (((client->_flag >> SOCKET_BUFFER_FLAG_PARENT_SOCKET_POS) & 0xff) == (uint8)_socket) {
				n += client->write(buffer, size);
			}
		}
	}
	return n;
}
```
to this

```
size_t WiFiServer::write(const uint8_t *buffer, size_t size)
{
	size_t n = 0;
	WiFiClient *client;

	for (int sock = 0; sock < TCP_SOCK_MAX; sock++) {
		client = WiFi._client[sock];
		if (client && client->_flag & SOCKET_BUFFER_FLAG_CONNECTED) {
			if (((client->_flag >> SOCKET_BUFFER_FLAG_PARENT_SOCKET_POS) & 0xff) == (uint8)(_socket + 1)) {
				n += client->write(buffer, size);
			}
		}
	}
	return n;
}
```
the socket needs to be added with 1 like its done in WiFiClient WiFiServer::available(uint8_t* status)

sandeepmistry, 2017-05-02:
@MarkusLange thanks for brining this up, I've opened pull request https://github.com/arduino-libraries/WiFi101/pull/162 with the fix you suggested.

sandeepmistry, 2017-05-08:
Closed via #162 which has been merged.

closed 2017-05-08 by sandeepmistry