arduino-libraries/WiFi101 #161: MKR1000: Crash of VNC/WiFI101 if another device connects to the WLAN

facchinm, 2017-05-02:
_Moved from https://github.com/arduino/Arduino/issues/6237_

_@DieterBurandt wrote:_

I use two Genuino MKR1000 (MR510PB), now Firmware 19.5.2, Arduino 1.8.2, samd: 1.6.14 and 1.6.15, WiFI101: 0.14.1, IDE: VS2017 + Visual Micro 1704.03, (also Atmel  Studio and Arduino IDE), DSL-Router: AVM Fritz!Box 7490 FW:06.83, Access point: AVM FRITZ!Powerline 546E FW: 06.50.

Two MKR1000 Boards:
1) One original bare one
2) One board with user hardware: RTC, SRAM, EEPROM, SD-card, a serial device. Web-pages using modified “Webduino“.

All attached devices are working well. I looked further on regarding my failing WLAN connections. The error occurs within a few Minutes and up to more then 4 hours. Then:

-  The loop is still executed. WiFi.status returns “WL_CONNECTED”.
-  Ping from the MKR1000 to my gateway returns forever “okay”.
-  Reading the internal Mac or reading the signal strength returns 0.
- The Led at Pin 6 is blinking in half second intervals that I  did not programmed.
-  The board is dead in respect to the WLAN. Trying to reconnect mostly fails. (But I don’t know how to reconnect correctly.)
-   After a failure, sometimes two power fail and reboots are required the get access to the WLAN again. The ATWINC1501 Model B firmware or the WiFi101 driver seems being crash completely.

I tested a lot to find any reproducible procedure to force the bug. I am sure there is more than one reason for the failure. I found one method to reproduce it:

If I connect one of our two laptops to the WLAN both of my MKR1000 fails exactly in the same moment. When you insert code in the loop that causes the LED to blink, you will see a short wait time of around a second when another device is connecting to the WLAN and then sometimes the connection of the MKR1000 fails and sometimes not. With my two laptops, the bug is well reproducible, but not with my mobile.  I also could see, that the second MKR1000 often crash when I connected the first one to the WLAN. (One of my MKR1000 is running my application, the other one is running the example code of “WiFiWebServer”.) On failure, after around a minute, the MKR1000 enters a low power mode even noLowPower is specified.

I tested with my two access points, I used  CCMP or TKIP, I changed a pin and did not use pin 7, I took power bank to supply the bare MKR1000 far away from electrical noise, nothing of that had any influence on the bug.

This failure may be specific to some access points/routers products. But keep in mind that the  AVM Fritz!Boxes have a market share of more than 50% in Germany and they sell their products in a lot of countries too. Other users also reported trouble with the MKR1000 loosing the WLAN connection already.

With this bug, all products using the ATWINC1501 and/or WiFi101 ares not usable for me. I spend a lot of time, but I must stop my project with this board. My project will clearly delayed by month.

At is worth looking at two points: 

1) The MKR1000 was extremely sensitive to electrical noise I think induced by my screen that was near by. If I did come near to the board with my hand (5 to 10 cm not touching) the MKR1000 mostly stopped working. After initializing all user pins properly the MKR1000 was stable. There will be a pin configured as input activated to handling interrupts.  (tested using FW: 19.4.4).  One should clarify which pins are used by the VNC and also if there are pins are not initialized properly . A user must know which of the pins can be used and which of them not. In contrast tp the documentation digital pin 7 seems not to be used by the VNC, but pin 6. But despite this pin 6 can be used by the user with restrictions.

2) If you try to connect to an access point (or a router) with an SSID and a key, WiFI101 uses CCMP or TKIP. In both cases „WiFi.encryptionType();“ returns ENC_TYPE_TKIP.  That is incorrect if CCMP is actually used. You can check this using „WifoInfoView“ from www.nirsoft.net on your Laptop and changing the mode of the access point. Other users got already confused and thought only TKIP is supported.

For security reason CCMP setting of the cipher mode must be supported without any fallback to TKIP.

(It would be very interesting to see if the above crash still exist, if you define the encryption method explicitly.)

facchinm, 2017-05-02:
The internal WINC1500 chip is routed through non-exposed pins, so it's very unlikely that the hang is provoked by the (user) pins configuration. Instead, the hang when connecting another PC to the network is far more interesting and should be investigated. @sandeepmistry have you ever seen something like this?

mamama1, 2017-05-02:
hi
try using esp32 or esp8266 for wifi. the non-counterfeits with good spi flash work really well if you yield() on a regular basis. robotdyn makes good devices (esp8266).
I'm desperate trying to make projects with the winc1500. I've thrown all my modules away, although they were about 20€ each.
regards

DieterBurandt, 2017-05-02:
Hi mamama1,
my MKR1000 is working very well with lots of decices being connected with the exception of that what I described above.
regards

DieterBurandt, 2017-05-02:
Hi facchinm,

on the WiFi library page "www.arduino.cc/en/Reference/WiFi101" there I found the following sentence: "Digital pin 7 is used as a handshake pin between the WiFi shield 101 and the underlying board, and should not be used." It is not clear for me, if this is correct also for the MKR1000. The very high sensitivity for electrical noise using the FW 19.4.4 was curious. Currently I cannot see this, but I use FW 19.5.2 now. I mentioned this, because it can help you when you looking for the bug.
regards

facchinm, 2017-05-02:
The pin7 reference is only related with Wifi101 shield because all the IOs must be user accessible; the MKR1000 implements everything internally and doesn't expose any of that pins :wink:

sandeepmistry, 2017-05-02:
>  @sandeepmistry have you ever seen something like this?

@facchinm no I haven't. This seems more like a firmware specific issue.

I would suggest opening a support ticket with Microchip/Atmel support: http://www.atmel.com/design-support/ - as the firmware is closed source and maintained my Microchip/Atmel.

@facchinm any other suggestions?

facchinm, 2017-05-03:
No suggestion in fact, the whole process should be sniffed via Wireshark or similar to get a better idea of what is going on on the network but, since it's the first report about this problem, it looks difficult to reproduce without the same router router and the same PCs NICs.
Said that, involving Atmel is probably the best idea

DieterBurandt, 2017-05-04:
Hi Martino,

It is strange for me, that this is a problem that only I have. I can see mails of some users reporting also trouble with the WiFi, but no one analyzed it so deeply.

This morning I switched off the WiFi-option "WiFi devices may communicate to each other" on the router and I switched off all other WiFi-devices I have. This gives a little different behavior: Sometimes the MKR1000 could recover automatically, but in effect the bug still exist.

I analyzed already using Wireshark on my desktop PC. In addition the router does have maintenance functions to log the WiFi activities displayed via Wireshark. I am not the specialist for the WiFi. I could not see any unusual packets and had trouble to understand the packages. It looks as I could not trace the low level WiFi calls. The old Ethernet Lan is much more my world.

I can give you Wireshark traces from the Fritz!Box 7490 router of type:

ath0, ath1, eoam, eth0, eth1, eth2, eth3, lan, ptm_vr9, tunl0, wasp, wifi0, wifi1 ,wlan,

AP2 (2.4 + 5 GHz, ath1) - Port 0    (or Port 1)

AP (2.4 GHz, ath0) - Port 0     (or  Port 1)

AP (2.4 GHz, ath0) - Port 0

HW (2.4 GHz, wifi0) - Port 0

WLAN Management Traffic - Port 0

(DTrace -D-s-m-i256-dect-dlc-c1-c2-c3-c4-c5-nt3-d2-d3)      (whatever this is)

I don't think, that the router can trace all of them in parallel. But more than 5 in parallel is definitely possible.

In addition, my none-bare MKR1000 is a system with external SRAM caches log data that are finally written to the SD-card. If I know what to trace, I can do it with a speed of up to 16MBit/s (SPI-clock).

I agree with you. I also expect a firmware bug.

When I look into the WiFi101 library sources I think the programmer had to struggle a lot to get it working. (Code like I wrote sometimes too.) The bug may have to do with buffer sizes, timing, and/or packet handshake. It would be interest to see if the bug still exist, if one can explicitly specify the WPA mode TKIP or CCMP (no automatic selection). If there is only one protocol supported, packages for the other one can be immediately ignored. Also the code will be simplified.

Best regards,

Dieter

sandeepmistry, 2017-05-10:
Hi @DieterBurandt,

Thank you for sharing the additional details.

Could you please open a Support Case with Atmel/Microchip here: http://www.atmel.com/design-support/

Then provide us with the case number. It would be great if you could also keep this Github issue in sync with the info provided via the support case. Thanks.

DieterBurandt, 2017-08-06:
Hi sandeepmistry,

After a lot mails to Atmel/Microsystems they found the reason for these crashes when I connect my laptop or a second my MKR1000 to my WiFI network. The reason is, that there send also a UDP package with a size of zero to the MKR1000. This causes the MKR1000 to crash due to a bug in the firmware 19.5.2 and oldder. This bug is fixed in the next firware 19.5.3 as I could test. Please give this information to your team and to your users.

(On testing your example AP_SimplwWebServer I must see, that this Web server is working only, when the  
Serial Monitor is active and can send log data to the connetcted PC. This is a bug on your side and should be well reproducable. You should verify and correct it.)

Best Regards,

Dieter Burandt

sandeepmistry, 2017-10-18:
Hi @DieterBurandt,

Thanks for the update.

> The reason is, that there send also a UDP package with a size of zero to the MKR1000. This causes the MKR1000 to crash due to a bug in the firmware 19.5.2 and oldder. This bug is fixed in the next firware 19.5.3 as I could test. Please give this information to your team and to your users.

Let's close this for now and track in #185, we should have a fix with the upcoming WINC1500 19.5.4 release.

closed 2017-10-18 by sandeepmistry