arduino-libraries/WiFi101 #163: WifiUDP.write truncates packets at 72 bytes

subunit, 2017-05-03:
Hello- I am using the Wifi101 library with an adafruit ATWINC1500 breakout board connected to an arduino Mega to send some data to a remote computer. After I was unable to process this information correctly, I checked using wireshark and observed that the packet messages are truncated to 72 bytes. I am able to replicate the issue with the following example code:

`#include <stdlib.h>
#include <WiFi101.h>
#include <WiFiUdp.h>

//WiFi variables
char ssid[] = "ssid"; 
char pass[] = "password";
int status = WL_IDLE_STATUS; 
unsigned int localPort = 2390;
char remoteIP[ ] = "10.0.0.2";
WiFiUDP Udp; 

void setup() {
 Serial.begin(9600);
  //Configure pins for Adafruit ATWINC1500 Breakout
  WiFi.setPins(32,22,23);

  while ( status != WL_CONNECTED) {
    status = WiFi.begin(ssid, pass);
    delay(10000);
  }

  Udp.begin(localPort);
}

void loop() {
  Udp.beginPacket(RD05IPAddress,6000);
  int messageSize = Udp.write("1.ABCDEFGHIJKLMNOPQRSTUVWXYZ.2.ABCDEFGHIJKLMNOPQRSTUVWXYZ.3.ABCDEFGHIJKLMNOPQRSTUVWXYZ.4.ABCDEFGHIJKLMNOPQRSTUVWXYZ",115);
  Serial.print("message bytes:");
  Serial.println(messageSize); 
  boolean packetdebug = Udp.endPacket();
  Serial.println(packetdebug);
}`

As you can see, I am attempting to write a 115 byte message. int messageSize evaluates as 72, the boolean packetdebug as 1 (packet is sent successfully), and I see the 72 byte truncated message at the remote computer on wireshark. It doesn't matter whether I use one or multiple Udp.write commands, the packet is never longer than 72 bytes. I have no idea why this should be the case. I can work around this for now, but I would really like to be able to send longer packets in the future. Can anyone suggest how to fix this?

gui77aume, 2017-05-10:
The workaround I use consists in editing WiFi101/src/socket/include/socket_buffer.h and changing
the SOCKET_BUFFER_MTU value (which is 64 for AVRs). The drawback is, of course, the memory consumption.

subunit, 2017-05-10:
Thanks gui77aume! I was able to resolve this by changing line 58:
`
#define SOCKET_BUFFER_UDP_SIZE					(SOCKET_BUFFER_UDP_HEADER_SIZE + 64u)`

from 64u to a higher value. I'm not sure why the arduino mega board I'm using was evaluating as a "LIMITED_RAM_DEVICE" rather than "ARDUINO", which would have given a much larger socket buffer. Thanks again.

sandeepmistry, 2017-05-10:
@facchinm @cmaglie any thoughts on increasing the buffer size for the Mega board since it has more RAM compared to other AVR boards like the Uno?

subunit, 2017-05-11:
Ah- I see my misunderstanding now, the Mega is __supposed__ to use the limited ram value. That's probably fine, but this is the kind of thing that desperately needs to be in the Wifi101 reference documentation on arduino.cc- an explicit statement of the maximum default packet size along with the workaround for changing it. The only material I had to go on was this github thread- [https://github.com/arduino-libraries/WiFi101/pull/59](url), which obviously an amateur like me is not equipped to parse for answers to "why are my packets limited to 72 bytes". Thanks for all the help!

sandeepmistry, 2017-05-11:
@akash73 could we get the WiFi101 reference pages updated for WiFiUdp please.

On AVR based boards, as discussed above, outgoing UDP packets are limited to 72 bytes in size currently. For non-AVR boards the limit is 1446 bytes.

sandeepmistry, 2017-10-18:
@akash73 ping, was this documented?

akash73, 2017-10-19:
yes, pushed.

closed 2017-10-19 by akash73