arduino-libraries/WiFi101 #164: WINC1500 XPLAINED PRO 19.5.2 firmware update SAMD21

Dragonzss1, 2017-05-04:
Is it possible to use this code to update my WINC1500 through the SAMD21? I need to update to 19.5.2 to stop the socket buffer full error

kbumsik, 2017-05-04:
Not sure how to do that using Arduino IDE, but you can do it using Atmel Studio IDE.
There is an example project that you can update a new firmware. Go to [Atmel's document, section 5, How to Download New ATWINC1500 Firmware](http://www.atmel.com/images/atmel-42417-getting-started-guide-for-atwinc1500wifi-using-samd21-xplained-pro_userguide.pdf)

Dragonzss1, 2017-05-04:
Thanks for the information, I finished the update and updated the drivers, but now the buffer gets full after 38 published messages instead of 7. Transferring at 1 publish or second. Was this firmware update suppose to fix this problem?

closed 2017-05-04 by Dragonzss1