arduino-libraries/WiFi101 #165: WINC1500 can publish and subscribe but cannot recieve

Dragonzss1, 2017-05-04:
So I can publish and subscribe to topics on my MQTT broker, but i cannot receive a published message from other users that publish on the same topic. Am i suppose to use interrupts  for this and if so how do i go about doing that?

Dragonzss1, 2017-05-05:
It seems as though the Interrupt pin on the WINC1500 never asserts. Even though the sever is saying that something has been sent back to the client. Is there a condition that needs to be meet before a interrupt is set? What i can infer from this is that the winc1500 client is simply not receiving anything

kbumsik, 2017-05-05:
As far as I know, Wifi101 library doesn't have MQTT client implementation. Which MQTT client and server are you using?

sandeepmistry, 2017-05-10:
@Dragonzss1 it's not clear if you are using this library.

Please provide an Arduino sketch and steps to reproduce the issue, then re-open this issue.

closed 2017-05-10 by sandeepmistry