arduino-libraries/WiFi101 #170: LGPLv2.1 License problem.

kbumsik, 2017-05-17:
Hi,
I was porting this library to my own board but I notice there is [a potential problem in licensing here](https://github.com/arduino-libraries/WiFi101/blob/master/src/socket/source/socket_buffer.c#L26).

```
 * 4. This software may only be redistributed and used in connection with an
 *    Atmel microcontroller product.
```

I am not sure but this copyright notice might not be incompatible with LGPLv2.1. And also this library is used on many other non-Atmel MCU board. What do you think about this?

sandeepmistry, 2017-05-23:
Hi @kbumsik,

We've updated the license in https://github.com/arduino-libraries/WiFi101/commit/6ad9880066141d278c0be21892615b682581b997 to remove this clause.

closed 2017-05-23 by sandeepmistry

kbumsik, 2017-05-23:
Oh, thank you for your response :)