arduino-libraries/WiFi101 #171: Latest Wifi101 + concurrent connection to SoftAp

melvinpmathew, 2017-05-22:
Just wanted to know if concurrent connections can be made to a SoftAP. I have one device connected to the SoftAP, but not able to connect the second device to the SoftAP. I can have only 1 device connection active at a time. 

I have tried with the latest code and also the workaround for the issue :https://github.com/arduino-libraries/WiFi101/issues/74, seems to be not the latest.

Is there anyway to achieve this? Any inputs will be great! @sandeepmistry

melvinpmathew, 2017-05-22:
Basically I need two devices connected to the SoftAP.

melvinpmathew, 2017-05-23:
Any inputs anyone? :)

sandeepmistry, 2017-05-23:
Hi @melvinpmathew,

This is a limitation of the WINC1500 firmware which is maintained by Microchip/Atmel.

Please open a support case with them to request this feature: http://www.atmel.com/design-support/

melvinpmathew, 2017-05-23:
Thanks a lot for the reply @sandeepmistry . Let me go ahead and open a support ticket with them.

rosenk, 2017-05-23:
@melvinpmathew That's why we should use Raspberry Pi Zero W -> way cheaper and much better software support. I just don't see the economical reasoning behind Arduino any more.

rosenk, 2017-05-23:
http://www.atmel.com/Images/Atmel-42420-WINC1500-Software-Design-Guide_UserGuide.pdf

> 8.3 Limitations

-  AP mode supports OPEN and WEP security only
-  **The AP can only support a single associated station. Further connect attempts will be rejected.**
-  Concurrency (simultaneous STA/P2P and AP mode) is not supported. Before activating the AP mode,
host MCU application should disable the mode currently running.

melvinpmathew, 2017-05-23:
Thanks @rosenk , I need to check into the Raspberry Pi Zero W. Seems like the way to go.

closed 2017-05-23 by melvinpmathew

rosenk, 2017-05-23:
I hear good things about Esp8266 and esp32 too.