arduino-libraries/WiFi101 #174: Tons of issues with WiFi101 lib on all platforms

liamnaddell, 2017-05-25:
I own a mac, a windows computer, a debian computer and an arch linux computer.
But most of the information comes from an arch linux computer. 
SPECS:
```
NodeMcu esp8266 connected via usb to arch linux computer.
arch linux runs arduino IDE on gnome3 and x11. I successfully compiled the blink program under file>examples.
I installed Wifi101 via the Include Libraries button on the IDE. 
```
I am a programmer, but not a c++ programmer. So I might just be doing something really stupid, but:
No matter what platform I am on, I will always get the same errors:

# 1st error:
```
Arduino: 1.8.2 (Linux), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

/root/Arduino/libraries/WiFi101/src/WiFiMDNSResponder.cpp:24:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

This can be solved by deleting the `avr/` part of the #include statement

# 2nd error:

```
Arduino: 1.8.2 (Linux), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

In file included from /root/Arduino/libraries/WiFi101/src/common/include/nm_common.h:45:0,
                 from /root/Arduino/libraries/WiFi101/src/driver/include/m2m_wifi.h:49,
                 from /root/Arduino/libraries/WiFi101/src/WiFi101.h:28,
                 from /root/Downloads/ConfiguredFirmata/ConfiguredFirmata.ino:12:
/root/Arduino/libraries/WiFi101/src/bsp/include/nm_bsp.h:110:23: error: conflicting declaration 'typedef long unsigned int uint32'
 typedef unsigned long uint32;
                       ^
In file included from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h:24:0,
                 from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:38,
                 from sketch/ConfiguredFirmata.ino.cpp:1:
/root/Arduino/hardware/esp8266com/esp8266/tools/sdk/include/c_types.h:30:29: error: 'uint32' has a previous declaration as 'typedef unsigned int uint32'
 typedef unsigned int        uint32;
                             ^
In file included from /root/Arduino/libraries/WiFi101/src/common/include/nm_common.h:45:0,
                 from /root/Arduino/libraries/WiFi101/src/driver/include/m2m_wifi.h:49,
                 from /root/Arduino/libraries/WiFi101/src/WiFi101.h:28,
                 from /root/Downloads/ConfiguredFirmata/ConfiguredFirmata.ino:12:
/root/Arduino/libraries/WiFi101/src/bsp/include/nm_bsp.h:133:22: error: conflicting declaration 'typedef long int sint32'
 typedef signed long  sint32;
                      ^
In file included from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h:24:0,
                 from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:38,
                 from sketch/ConfiguredFirmata.ino.cpp:1:
/root/Arduino/hardware/esp8266com/esp8266/tools/sdk/include/c_types.h:33:29: error: 'sint32' has a previous declaration as 'typedef int sint32'
 typedef signed int          sint32;
                             ^
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

This can be fixed by deleting some typedef's in the file
src/bsp/include/nm_bsp.h

If you delete lines 133 and 110, then you end up with the 3rd error:

```
Arduino: 1.8.2 (Linux), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

In file included from /root/Arduino/libraries/WiFi101/src/common/include/nm_common.h:45:0,
                 from /root/Arduino/libraries/WiFi101/src/socket/include/socket.h:53,
[... 138 lines cut ...]
  ^
/root/Arduino/libraries/WiFi101/src/driver/include/m2m_types.h:2043:2: error: 'uint32' does not name a type
  uint32 u8NcdUpgradeVersion;
  ^
In file included from /root/Arduino/libraries/WiFi101/src/WiFiClient.cpp:22:0:
/root/Arduino/libraries/WiFi101/src/driver/include/m2m_periph.h:404:38: error: 'uint32' was not declared in this scope
 NMI_API sint8 m2m_periph_pullup_ctrl(uint32 pinmask, uint8 enable);
                                      ^
/root/Arduino/libraries/WiFi101/src/driver/include/m2m_periph.h:404:60: error: expected primary-expression before 'enable'
 NMI_API sint8 m2m_periph_pullup_ctrl(uint32 pinmask, uint8 enable);
                                                            ^
/root/Arduino/libraries/WiFi101/src/driver/include/m2m_periph.h:404:66: error: expression list treated as compound expression in initializer [-fpermissive]
 NMI_API sint8 m2m_periph_pullup_ctrl(uint32 pinmask, uint8 enable);
                                                                  ^
/root/Arduino/libraries/WiFi101/src/WiFiClient.cpp: In member function 'int WiFiClient::connect(IPAddress, uint16_t, uint8_t, const uint8_t*)':
/root/Arduino/libraries/WiFi101/src/WiFiClient.cpp:132:16: error: 'struct in_addr' has no member named 's_addr'
  addr.sin_addr.s_addr = ip;
                ^
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

I have no idea how to solve the following.

This is where I need help.

liamnaddell, 2017-05-25:
I have been following this guide:
https://gobot.io/documentation/platforms/esp8266/
and the file is here:
```
*
 * ConfiguredFirmata.ino generated by FirmataBuilder
 * Thu May 25 2017 14:29:28 GMT-0400 (EDT)
 */

#include <ConfigurableFirmata.h>

// uncomment to enable debugging over Serial (9600 baud)
//#define SERIAL_DEBUG
#include "utility/firmataDebug.h"

#include <WiFi101.h>
#include "utility/WiFiClientStream.h"
#include "utility/WiFiServerStream.h"

#define WIFI_MAX_CONN_ATTEMPTS 20

#define NETWORK_PORT 3030

char ssid[] = "redacted";
char wpa_passphrase[] = "redacted";

WiFiServerStream stream(NETWORK_PORT);

int connectionAttempts = 0;
bool streamConnected = false;

#include <DigitalInputFirmata.h>
DigitalInputFirmata digitalInput;

#include <DigitalOutputFirmata.h>
DigitalOutputFirmata digitalOutput;

#include <AnalogInputFirmata.h>
AnalogInputFirmata analogInput;

#include <AnalogOutputFirmata.h>
AnalogOutputFirmata analogOutput;

#include <Wire.h>
#include <I2CFirmata.h>
I2CFirmata i2c;

#include <FirmataExt.h>
FirmataExt firmataExt;

#include <AnalogWrite.h>

#include <FirmataReporting.h>
FirmataReporting reporting;

void systemResetCallback()
{
  for (byte i = 0; i < TOTAL_PINS; i++) {
    if (IS_PIN_ANALOG(i)) {
      Firmata.setPinMode(i, ANALOG);
    } else if (IS_PIN_DIGITAL(i)) {
      Firmata.setPinMode(i, OUTPUT);
    }
  }
  firmataExt.reset();
}

void hostConnectionCallback(byte state)
{
  switch (state) {
    case HOST_CONNECTION_CONNECTED:
      DEBUG_PRINTLN("TCP connection established");
      break;
    case HOST_CONNECTION_DISCONNECTED:
      DEBUG_PRINTLN("TCP connection disconnected");
      break;
  }
}

void printWiFiStatus()
{
  if (WiFi.status() != WL_CONNECTED) {
    DEBUG_PRINT("WiFi connection failed. Status value: ");
    DEBUG_PRINTLN(WiFi.status());
  } else {
    DEBUG_PRINTLN("Board configured as a TCP server");

    DEBUG_PRINT("SSID: ");
    DEBUG_PRINTLN(WiFi.SSID());

    DEBUG_PRINT("Local IP Address: ");
    IPAddress ip = WiFi.localIP();
    DEBUG_PRINTLN(ip);

    DEBUG_PRINT("Signal strength (RSSI): ");
    long rssi = WiFi.RSSI();
    DEBUG_PRINT(rssi);
    DEBUG_PRINTLN(" dBm");
  }
}

void initTransport()
{
  // IMPORTANT: if SERIAL_DEBUG is enabled, program execution will stop
  // at DEBUG_BEGIN until a Serial conneciton is established
  DEBUG_BEGIN(9600);
  DEBUG_PRINTLN("Attempting a WiFi connection using the WiFi 101 library.");

  DEBUG_PRINTLN("IP will be requested from DHCP ...");

  stream.attach(hostConnectionCallback);

  DEBUG_PRINT("Attempting to connect to WPA SSID: ");
  DEBUG_PRINTLN(ssid);
  stream.begin(ssid, wpa_passphrase);

  DEBUG_PRINTLN("WiFi setup done.");

  while(WiFi.status() != WL_CONNECTED && ++connectionAttempts <= WIFI_MAX_CONN_ATTEMPTS) {
    delay(500);
    DEBUG_PRINT(".");
  }

  printWiFiStatus();

  Firmata.begin(stream);
}

void initFirmata()
{
  Firmata.setFirmwareVersion(FIRMATA_FIRMWARE_MAJOR_VERSION, FIRMATA_FIRMWARE_MINOR_VERSION);

  firmataExt.addFeature(digitalInput);
  firmataExt.addFeature(digitalOutput);
  firmataExt.addFeature(analogInput);
  firmataExt.addFeature(analogOutput);
  firmataExt.addFeature(i2c);
  firmataExt.addFeature(reporting);

  Firmata.attach(SYSTEM_RESET, systemResetCallback);
}

void setup()
{
  initFirmata();

  initTransport();

  systemResetCallback();
}

void loop()
{
  digitalInput.report();

  while(Firmata.available()) {
    Firmata.processInput();
  }

  if (reporting.elapsed()) {
    analogInput.report();
    i2c.report();
  }

  stream.maintain();
}
```

sandeepmistry, 2017-05-29:
Hi @liamnaddell,

I'm going to close this for now, as it doesn't really look like you want to use the WiFi101 library on the ESP8266 (because you need some additional hardware).

I think you are just not selecting the right board in the drop down:
[image]

closed 2017-05-29 by sandeepmistry