arduino-libraries/WiFi101 #177: lockup issue

tonton81, 2017-06-11:
dont know if this is a teensy 3.5 issue or not, im able to make the adafruit winc1500 work on one 3.5 but not another, on both SPI0 and SPI1. The problem is when it doesn't work, it will lock up the entire mcu code
(block forever)

try to boot the MCU without the winc1500 connected, you will see a complete lockup.
If the Winc1500 ever fails, this can lead to disastrous results

sandeepmistry, 2017-10-18:
Hi @tonton81, What version of the WiFi101 library are you using? Could you also please share a minimal sketch that locks up?

tonton81, 2017-10-18:
latest one off this github, using a teensy 3.5, its been a long while so ill have to look for the code in my history sketches

sandeepmistry, 2017-10-19:
Hi @tonton81,

I suggest you contact Adafruit customer support, we cannot reproduce the issue with a MKR1000 or WiFi101 shield.

If it turns out to be an issue with the library, please re-open the issue with more details.

closed 2017-10-19 by sandeepmistry