arduino-libraries/WiFi101 #178: Wifi hangs on post/get calls after 0.14.0 release

oakmusic, 2017-06-14:
Hi,

I'm using ArduinoHttpClient library (https://github.com/arduino-libraries/ArduinoHttpClient) to make post/get calls. And everything went right until Wifi101 0.13.0 release.

From 0.14.0 and later, it hangs after 3 correct calls. The first three calls (post or get) work fine, but it hangs at the fourth one.

I've tried 0.14.0, 0.14.1, 0.14.2, and 0.14.3 but all of them breaks. I'm still using 0.13.0.

klausj1, 2017-06-16:
Hi,

I also struggled with some issues, and created a version of the Wifi-Library with a lot of logging to the serial port (based on 0.14.3).

You could try this version [here](https://github.com/klausj1/WiFi101), and post the output. Preferably using [realterm](https://sourceforge.net/projects/realterm/), and turn on the human readable timestamp format so that each row starts with the current date/time. If your sketch hangs, wait at least for 2 minutes, there are long timeouts.

And please post your complete sketch, make sure that the logging of the ArduinoHttpClient is turned on (add `#define LOGGING 1` to your sketch), and put some logging in your sketch to see what happens there and where it hangs as well.

Regards,
Klaus

jasontwinters, 2017-10-02:
I can also confirm that something changed, in a negative way, with the release of 0.14.  The ThingSpeak library stopped working correctly with the release of 0.14.x.  Downgrading to 0.13.0 fixes the issue.

hoblins, 2017-11-28:
Unfortunately I also have to confirm that issue for WINC1500 firmware 19.5.4 and WiFi101 0.14.5.
Bizzarely, it occured from one day to the next without INTENTIONALLY changing the code on the microcontroller. So maybe I changed something by mistake or  maybe there were changes to php and its libs on the server side affecting the behaviour of the php script.
My workaround is a) to send the request directly without ArduinoHttpClient, b) add a delay between sending the request and reading the answer and c) add a .stop() before the .connect(). Maybe its also working with ArduinoHttpClient and this additions.
OLD:
```cpp
WiFiSSLClient wifiClient;

if (wifiClient.connect(server, serverPort)) {
  wifiClient.println("POST /script.php HTTP/1.1");
  ...
  while (wifiClient.available()) {
    char c = wifiClient.read();
    Serial.write(c);
  }
}
```
WORKAROUND:
```cpp
WiFiSSLClient wifiClient;

wifiClient.stop();
if (wifiClient.connect(server, serverPort)) {
  wifiClient.println("POST /script.php HTTP/1.1");
  ...
  delayMicroseconds(250000);
  while (wifiClient.available()) {
    char c = wifiClient.read();
    Serial.write(c);
  }
}
```
This does not completely fix the posts but instead of about 3 requests there are many (>100) requests possible. After a reboot sending requests works again.

Rocketct, 2018-07-30:
Hi to all, today i have tested the version 0.15.1 of the WiFi101 library, whit the firmware 19.5.4, running the examples DweetPost https://github.com/arduino-libraries/ArduinoHttpClient/tree/master/examples/DweetPost  and DweetGet https://github.com/arduino-libraries/ArduinoHttpClient/tree/master/examples/DweetGet of ArduinoHTTPclient library, the two example have reach 300 connections and now are still run, the problem seems fixed, i close this issue.

closed 2018-07-30 by Rocketct