arduino-libraries/WiFi101 #179: Updating firmware fails after erasing original firmware

armanafghani, 2017-06-14:
I'm currently trying to set up the ATWINC1500 breakout board with my Adafruit Bluefruit Feather 32u4. I connected the two boards via SPI, and first ran the CheckWiFi101FirmwareVersion program. This ran perfectly fine, and stated I needed to update from 19.4.4 to 19.5.2. I then proceeded to run the UpdateFirmware program to do so.

When running UpdateFirmware program, I selected WINC1501 Model B (19.5.2) and clicked "update firmware" in the WiFi101Firmware/Certificates updater window, and I got an error message back saying "error while writing to flash memory" 

Now every time i run the program again, I get an error saying "Programmer not responding'

Unfortunately during the process of updating firmware the first time, the updater window logged that it was erasing my firmware, now the CheckFirmware program shows that no shield is connected.

Any ideas on how to fix?

armanafghani, 2017-06-14:
Here is the exact error that i'm getting
`java.lang.Exception: Error while writing flash memory.
	at cc.arduino.plugins.wifi101.flashers.java.FlasherSerialClient.writeFlash(FlasherSerialClient.java:133)
	at cc.arduino.plugins.wifi101.flashers.java.JavaFlasher.updateFirmware(JavaFlasher.java:86)
	at cc.arduino.plugins.wifi101.UpdaterImpl$3.run(UpdaterImpl.java:138)
`

sandeepmistry, 2017-10-18:
Hi @armanafghani, is this still happening? Are you following the guide here? https://www.arduino.cc/en/Tutorial/FirmwareUpdater

Please open an issue in the following repo if it still occurs: https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/issues - we've only tested with the MKR1000 and WiFi101 shields.

closed 2017-10-18 by sandeepmistry