arduino-libraries/WiFi101 #181: WiFi Library does not support Watchdog

klausj1, 2017-06-16:
Hi,

if you want to activate the watchdog in your project, you cannot use the WiFi Library (or probably better the other way round).

```
  Serial.print("Enabling WDT ... ");
  wdt_enable(WDTO_8S);
  Serial.println("Done");

```

The watchdog has to be resetted at least every 8 seconds, and there are a lot of timeouts in the WiFi Library which are longer then these 8 seconds.

In my opinion it would be fine to support the watchdog in the WiFi Library. Would it make sense to simply call `wdt_reset();` in all these loops?

What do you think?

Regards,
Klaus

tonton81, 2017-06-16:
if you were threading you wouldnt have this issue :)

ericbaril72, 2017-07-17:
I do not think this is an issue with Wifi101... many libraries have long blocking functions.
They have to be processed by the application itself.

I use a timer Interrupt to call wd.reset() for wifi ( long & blocking ) related functions.   
I also make the wd.reset() conditional to a maximum timeout ( different TimeOut values for different functions )

examples:
SDcard --> sd.ls() can take a while on large capacity ...

Init yourself a timer ( shorter then your required WD period ) and call a wd.reset() only on those specific long blocking functions.

in my TC3_handler()  runnning at 100ms:
    if ((inDirlist)	&& millis()<dirlistWDtimeout)                                             wd.reset();
    if ((inWifiPost)	&&  millis()<WifiPostWDtimeout)                                wd.reset();
    if ((inWifiFileDownload)	&&  millis()<WifiFileDownloadWDtimeout) wd.reset();

tonton81, 2017-07-31:
or leave it as is and throw your wifi calls in a second thread, no need to waste timers :)

PKGeorgiev, 2017-07-31:
@ericbaril72 I have two questions:
1) Does the timer work during a WiFi blocking call?
2) What's the point of resetting WD via timer if you want WD to watch after your code that is using WiFi blocking calls? For example I had a project that checks if a TCP port is opened. The call to .Connect() takes more than 8 seconds if the port is not responding. Which is more than WD can wait (as @klausj1 mensioned). How to avoid this for example?

ericbaril72, 2017-07-31:
@pkgeorgiev,

.connect() is a very good example as it can block up to 30 or 120 seconds ( I did not update my wifi library in a while.. ).

I just finished modifying .connect() to make it  non-blocking as I need the SPI port during that time ( up to 120seconds).

Wifi library interrupt triggers a flag which lets lower priorities interrupt still be usable during blocking functions.
I don't recall where TC stands in the priority list.
1- to answer your question , even if wifi did have higher priority, it is not disabling everything else during the whole blocking call.

2- only kicking the dog for exceptions...
Loop() does the standard WD.rset() kicking.

I currently use the RTC interrupt ( every seconds ) rather then a TC /tcc timer.

Just before my wifi.connect() , I set a flag and a "time stamp".

My RTC interrupt handler only "kick the dog"
If the flag is SET, and that current time is smaller then "time stamp"+ "some-timeout-for-wifi-connect".

This way, I keep kicking the dog but with a different wd-period-for-wifi-connect.

SD library also has blocking functions...
Example: sd.ls( recursive-flag)
Can keep listing for a long time when you have many files.

So having flags&timestamp lets you have different WD periods.

Imagine you send a large file. Wifi.connect() is done and your code does lots of while(!eof){client.write(file.read()}

In the end, all of your different internet access calls would require WD.reset() calls.
Example :   File-upload(), file-download(), msg-post(), msg-get(), get-network-time().

I set different flags and let my RTC interrupts kick the dog on long-duration-calls().
Otherwise my main-loop kicks the dog.

Am not at my computer for another week but can answer without code-snippets until then...

Which processor are your using ? I use the SAMD21 but I tried answering as if you were using a arduino UNO.

Am thinking of working on a wifi_library_fork to make it fully async but am expecting religious police coding backlash by imposing one way rather then some-other-way.
It's getting awefully close to threading.

-imposing freeRTOS does increase footprint and limits the userbase.
-offering my code might be too restrictive for anyone else's projects.

Let me know what you think

tonton81, 2017-08-09:
threading like RTOS, i havn't used threads on avr before but my first and only attempt was TeensyThreads on Teensy ARM boards, you can set up to 8 threads to run in parallel with timeslices as low as 1uS! this alone had me run 2 displays run in parallel while having SPI bus run it's own code without speed loss. visually with 2 displays you WILL see a difference, and to simulate delayed code in one thread just add delay(5000) or whatever, the other threads will keep going while your library (wifi101) takes its own sweet time.

sandeepmistry, 2017-10-18:
Hi @klausj1,

> In my opinion it would be fine to support the watchdog in the WiFi Library. Would it make sense to simply call wdt_reset(); in all these loops?

Those watchdogs are AVR only, so I don't think it makes sense.

As @ericbaril72 mentioned above in https://github.com/arduino-libraries/WiFi101/issues/181#issuecomment-315788811 maybe Arduino libraries are blocking and not watch dog compatible.

klausj1, 2017-10-19:
Thanks, I understand.

closed 2017-10-19 by klausj1