arduino-libraries/WiFi101 #182: WiFiClient::write can last forever

klausj1, 2017-06-16:
Hi,

the method  WiFiClient::write does not have a timeout, and in some cases (I do not know when this happens), it can last forever.

I used the WiFi-Library from [here](https://github.com/klausj1/WiFi101/blob/master/src) (its a 0.14.3 with additional logs, and some minor changes), and get this logfile (only in very rare cases): 
```
"15.06.2017 15:41:51",Host: WiFiClient::write: Sending Buffer took more than 5 seconds!!! Duration: 60000
"15.06.2017 15:42:51",!!!IS_CONNECTED: 1
```
So I think the method should get a timeout.

Regards,
Klaus

hoblins, 2017-08-21:
Hello,
I'm using the Feather M0 Wifi as a sensor node. The node sends some sensor values every minute to a server with an HTTP post request. Unfortunately not perfectly reproducible, the board regularly (couple of hours up to 2-3 days) hanged during data transfer (transfer LED turned on).
The patch from @klausj1 reduced the error to almost zero.
Best

sandeepmistry, 2017-10-18:
Hi @klausj1,

Could you please submit a pull request for the specific changes to fix the issue. Thanks.

klausj1, 2017-10-19:
OK, I will do, but I will need some time for this.

klausj1, 2017-11-25:
Hi @sandeepmistry,

the pull request has been created, I hope everything is OK.

Klaus

sandeepmistry, 2018-01-02:
Closing this since PR #204 has be merged, which based on @klausj1's feedback has resolved this.

closed 2018-01-02 by sandeepmistry