arduino-libraries/WiFi101 #183: WiFi 101 TCP read multiple bytes issue on MKR1000

Sudeshna19, 2017-06-23:
I have MKR1000 configured as server. I am using IDE 1.8.1. In my code I want to receive say 2 bytes. The client sends 1 byte at a time. Say client sends b first and then A ideally server should read bA after the second byte sent from client. However, I notice the second value getting corrupted. I am pasting the code below:
```
#include <SPI.h>
#include<WiFi101.h>

int configureSuccess = WL_IDLE_STATUS;
char ssid[] = "xyz";
char pass[] = "*******";
WiFiServer server(20000);

static uint8_t alreadyConnected = 0;

void setup() {
Serial.begin(9600);
Serial1.begin(28800);
Serial.print("Attempting to connect to SSID: ");
Serial.println(ssid);
while ( configureSuccess != WL_CONNECTED) {
configureSuccess = WiFi.begin(ssid, pass);
// delay 10000
unsigned long startMillis = millis();
while(millis()- startMillis < 10000)
{
};
}
server.begin();

  for(int trialcount=0;trialcount<5;trialcount++)
  {
      if(Serial)
      {
          break;
      }else
      {
          unsigned long startMillis  = millis();
    while(millis()- startMillis < 1000)
                    {
                    };
      }
  }
  
  if (configureSuccess==WL_CONNECTED)
   { 
     IPAddress ip = WiFi.localIP();
     Serial.print("<<<IP address: ");
     Serial.print(ip);  
     Serial.println(">>>");            
  }
 else
 {
     Serial.println("<<< IP address :Failed to configure. >>>");
 }
}

void loop() {
// wait for a new client:
int avlBytes = 0;
int sizeReceived = 0;
int data2[2];

Serial1.println("loop");
WiFiClient client1 = server.available();
if (client1){
if(!alreadyConnected){
client1.flush();
alreadyConnected++;
Serial1.println("Flushing");
}
avlBytes = client1.available();
if(avlBytes >= 2){
sizeReceived= client1.read(data2,2);
Serial1.println("data_s");
Serial1.println(data2[0]);
Serial1.println(data2[1]);
Serial1.println("size rcvd");
Serial1.println(sizeReceived);
}
else{
Serial1.println("rcvd");
Serial1.println(avlBytes);
}
}
else {
alreadyConnected = 0;
Serial1.println("No client");
}
unsigned long startMillis = millis();
while(millis()- startMillis < 1000)
{
};
}
```

If client send C first then rcvd 1 is printed. If client then sends A then data_s 65 101 is printed. After this rcvd = 0 as nothing is sent. Next time if I send ABC then data_s 65 66. Then rcvd 1 is printed. If I now send D the data_s 67 121 is printed. Then I send A followed by A then I see data_s 97 66 is printed.

sandeepmistry, 2017-10-18:
Hi @Sudeshna19,

Unfortunately the sketch you provided does not compile, the line: `sizeReceived= client1.read(data2,2);` has an error. I think `int data2[2];` should be changed to `byte data2[2];`.

Things seem to be working after I make this change, are you sure you uploaded the right sketch? Please re-open with the exact sketch you are using.

closed 2017-10-18 by sandeepmistry

Sudeshna19, 2017-10-31:
Hi @sandeepmistry 

I agree the int data[2] was a typo. It is actually uint8_t data[2] in my sketch. I tried the attached sketch on IDE 1.8.1 with WiFi101 0.14.3. I sent 'B' first and then 'A' through TCPClient and received 65, 98 on Server. Then I sent 'c' followed by 'a' from Client and received 97, 98 in Server.  I have attached the sketch and serial monitor output.

You need to data one by one and not together from the Client. Say you type B  then click Send, type A then click Send. I tried on Windows10. I tried with an Android App as TCP Client and a Windows application for TCP Client for this issue. Both result in this issue.
[Server_multibyte_issue.txt](https://github.com/arduino-libraries/WiFi101/files/1430117/Server_multibyte_issue.txt)

[WiFi_multi.zip](https://github.com/arduino-libraries/WiFi101/files/1430121/WiFi_multi.zip)