arduino-libraries/WiFi101 #186: Issue with Sparkfun SAMD21 Mini Breakout

xristod, 2017-08-03:
Dear All,
I have successfully made Adafruit ATWINC1500 (https://www.adafruit.com/product/2999 ) working with Arduino Uno and Adafruit Pro Trinket but unfortunately I am experiencing problems with Sparkfun SAMD21 Mini Breakout (https://www.sparkfun.com/products/13664). The code sticks at 
// Check for the presence of the shield
  SerialUSB.print("WiFi101 shield: ");
and doesn't go further. I have noticed that a few other people had the same problem and probably the reason is the bootloader (https://forum.arduino.cc/index.php?topic=355532.0 , https://forums.adafruit.com/viewtopic.php?f=53&t=100377&hilit=atwinc1500+zero&start=15#p510593).
I have connected the SAMD21 to a SPI Real-time clock and works fine. So the problem is with the Wifi module.
Before I buy a debugger to burn the latest bootloader could you please advise me if there is any solution within the library?
Best,
George

ericbaril72, 2017-08-03:
Could you start with the minimum code to produce your problem ?

xristod, 2017-08-03:
Hello,
Thank you very much for your reply.
I just tried the version 0.50 and it prints "WiFi shield not present" so it goes beyond 
>> if (WiFi.status() == WL_NO_SHIELD)
So perhaps I have to find the latest version of Wifi101 that works with my SAMD21.

xristod, 2017-08-03:
I tried again the version 0.5.0 and it is not working again.
I uploaded the latest version without connecting the Wifi module but it doesn't print "NOT PRESENT".
It prints the following and sticks there ....
//----------------------------
WiFi101 firmware check.

WiFi101 shield: 
//----------------------------
Any help will be really appreciated.
Best,
George

melvinpmathew, 2017-08-03:
Did you try with a different board? Just to be sure. I have seen this issue with 1 out of 5 boards I have.

xristod, 2017-08-03:
Thank you for your reply. Unfortunately I've got only one. Perhaps I should order one directly from Sparkfun asking the latest version.

xristod, 2017-09-12:
Dear All,
i just tried again the CheckWifi101FirmwareVersion.ino and works fine with UNO but it fails again with Sparkfun SAMD21 Mini Breakout. I noticed that the problem is at the line "ret = m2m_wifi_init(&param);" at WiFi.cpp and I found that function at driver\source\m2m_wifi.c file.
Any help will be really appreciated!
Best,
George

sandeepmistry, 2017-10-18:
@xristod what pin outs are you connecting?

https://www.arduino.cc/en/Reference/WiFi101SetPins can be used to override the default pins which are 5 for reset, 7 for INTN, 10 for CS.

I would also suggest you use the latest version of this library.

xristod, 2017-10-19:
Hello Sandeep,
Thank you very much for your reply. 
I am connecting pins 11,12,13 (https://cdn.sparkfun.com/assets/learn_tutorials/4/5/4/graphicalDatasheet-Mini.pdf) to MOSI,MISO,SCK of the Adafruit ATWINC1500 breakout and 10,9,8 to CS,IRQ,RST respectively (WiFi.setPins(10,9,8);) while EN is connected to 3.3V.
I have also installed the 0.14.4 version of the Wifi101 library.
I have also checked that the wires are well connected using a multimeter.
Unfortunately it still does not work.
Any help will be really appreciated!
Best,
George

sandeepmistry, 2017-10-19:
Hi @xristod,

I suggest you contact Adafruit and/or Sparkfun customer support, we cannot reproduce the issue with a MKR1000 or WiFi101 shield.

If it turns out to be an issue with the library, please re-open the issue with more details.

closed 2017-10-19 by sandeepmistry