arduino-libraries/WiFi101 #188: SSL client connect fails after WiFi.hostByName on Adafruit Feather M0

JetstreamRoySprowl, 2017-08-07:
On Adafruit Feather M0, calling WiFi.hostByName prior to connect will cause the ssl connect call to fail. Attached is a failure case with the interesting code on line 88
[hostByName.zip](https://github.com/arduino-libraries/WiFi101/files/1203763/hostByName.zip)
. To run the failure case, add an iot_configs.h file containing the SSID and password for a wifi network.

The url in the attached code may be used for troubleshooting this problem; I'll keep it live for a while.

sandeepmistry, 2017-10-18:
Hi @JetstreamRoySprowl,

I can't compile the attached sketch, as `iot_configs.h` is missing.

If I try with the following sketch based on the bundled example things work:

```arduino
/*
  This example creates a client object that connects and transfers
  data using always SSL.

  It is compatible with the methods normally related to plain
  connections, like client.connect(host, port).

  Written by Arturo Guadalupi
  last revision November 2015

*/

#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "ssid"; //  your network SSID (name)
char pass[] = "pass";    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;            // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
//IPAddress server(74,125,232,128);  // numeric IP for Google (no DNS)
char server[] = "royspr-m07f130.azure-devices.net";    // name address for Google (using DNS)

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
WiFiSSLClient client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }
  Serial.println("Connected to wifi");
  printWiFiStatus();

  Serial.println("\nStarting connection to server...");
  // if you get a connection, report back via serial:
  IPAddress ip;
  WiFi.hostByName(server, ip);
  if (client.connect(server, 443)) {
    Serial.println("connected to server");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.1");
    client.println("Host: www.google.com");
    client.println("Connection: close");
    client.println();
  }
}

void loop() {
  // if there are incoming bytes available
  // from the server, read them and print them:
  while (client.available()) {
    char c = client.read();
    Serial.write(c);
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting from server.");
    client.stop();

    // do nothing forevermore:
    while (true);
  }
}

void printWiFiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

Could you please try this sketch and re-open if you still have issues with it? Thanks.

closed 2017-10-18 by sandeepmistry

sravani36, 2018-11-16:
Hi Sandeep,
I tried your code and used my IoTHub HostName. But, It is still throwing me error of IoTHub Not Found.
Details of Error, which I got in my serial monitor.
Starting connection to server...
connected to server
HTTP/1.1 401 Unauthorized
Content-Length: 152
Content-Type: application/json; charset=utf-8
Server: Microsoft-HTTPAPI/2.0
iothub-errorcode: IotHubNotFound
Date: Fri, 16 Nov 2018 16:58:12 GMT
Connection: close

{"Message":"ErrorCode:IotHubNotFound;Unauthorized","ExceptionMessage":"Tracking ID:b09fa83ccb6e4354b30c8b81e307b99e-G:11-TimeStamp:11/16/2018 16:58:12"}
disconnecting from server.
Can you help to resolve this?
Shravani