arduino-libraries/WiFi101 #190: Client Certs and private key

ShridharK91, 2017-09-09:
Hi,

I am trying to connect Arduino Pro M0 and ATWINC1500 to AWS IOT and it requires client certificates and private key to connect.

So I flashed the certs and key as per ATMEL documentation, but I don't know how to use the certs to connect to AWS IOT. Does the wifi101 library use the certs and key to connect and communicate with AWS server in any way?

With IBM which required no client certs and key, it was pretty simple to just use the WifiSSLClient example and connect to their server, but with AWS IOT, its not that simple and I wanted to know if it is this library that is preventing connection to AWS IOT or is it the ATWINC1500 firmware limitation or something else? 

Note: I am using the latest ATWINC1500 firmware 19.5.2 which supports mutual authentication and have flashed the right SSL, Client certificates and private key required to connect to my thing in AWS IOT.

broddo, 2017-09-22:
I'm trying to do something similar at the moment so I'll let you know how I get on. I've tried searching for documentation regarding the use of the ATWINC1500 for mutual authentication but the only thing I found was a single line in the release notes of the 19.5.2. Are you absolutely sure it supports it natively? Do you have any links to examples? 

I was going to try and do mutual auth using the ATECC508a cryptoauth module (to generate the client side certs). The benefit being that the private keys are stored securely. I think this would involve somehow intercepting the HTTPS handshake once the server responds looking for client cert - so a lot of work! 

I'll try loading the certs directly onto the WINC1500 and see what happens. My server is a custom one (not AWS). 

Anyway, regardless of the outcome, I recommend that you should look into the ATECC508a as that was built specifically for AWS IOT mutual auth.

sandeepmistry, 2017-10-18:
Hi @ShridharK91,

> So I flashed the certs and key as per ATMEL documentation, but I don't know how to use the certs to connect to AWS IOT. Does the wifi101 library use the certs and key to connect and communicate with AWS server in any way?

Do you have a link to the documentation, it should work transparently.

sandeepmistry, 2018-01-02:
I'm closing this for now due to lack of feedback, please re-open with requested info if you are still interested.

closed 2018-01-02 by sandeepmistry

jhoward1a, 2018-01-04:
Very interested in this, has anyone made any developments? 

As I understand it WiFi 101 seems to only download certificates from a given domain whereas for AWS IoT you need the ability to assign custom ones. Working on a few different methods but no luck yet.

robertpyke, 2018-03-03:
I'm also interested in this issue. I believe the issue is that you need to sign the requests with the cert provided by AWS iot. I don't see any way to specify the client cert.