arduino-libraries/WiFi101 #194: A new firmware (v19.5.2 -> v19.5.4): KRACK vulnerability and some TLS fix.

kbumsik, 2017-10-17:
Hi,
I am currently working on [WINC1500 Wi-Fi driver for an RTOS](https://github.com/RIOT-OS/RIOT/pull/7433) (not related to Arduino though). I just got a news that [Microchip released a new firmware](http://www.microchip.com/design-centers/wireless-connectivity/embedded-wi-fi/wpa2-protocol-vulnerability), responding to [KRACK](https://www.krackattacks.com/) issue ([VU\#228519](https://www.kb.cert.org/vuls/id/228519)). This also includes some issues of WINC1500 firmware, including TLS problems.

This new firmware seems quite important. Are you planning to add this firmware to the WiFi101 firmware uploader?

kbumsik, 2017-10-18:
The discussion of this issue has been done in https://github.com/arduino/Arduino/issues/6834 . And I will close this issue when https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/pull/11 is merged.

sandeepmistry, 2017-10-18:
I've also created https://github.com/arduino-libraries/WiFi101/pull/195 to update the latest model B firmware check once the new firmware updater has been released.

kbumsik, 2017-10-18:
Thank you very much for quick response. Out of curiosity, will it be available after the merge or next release of Arduino IDE/WiFi101 library?

sandeepmistry, 2017-10-18:
Very likely to be in the next IDE release if all testing goes well.

sandeepmistry, 2017-10-19:
The PR to include the new version of the firmware updater (and firmware release) in the IDE can be here: https://github.com/arduino/Arduino/pull/6840

PR #195 (2017-10-20): Bump latest model B version to19.5.4
To be merged once https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/pull/11 is merged and released.

Resolves #194.
closed 2017-10-20 by cmaglie