arduino-libraries/WiFi101 #196: with wifi101 latest and arduino wifi board fw 19.4.4 , a .get() returns a bunch of leading FFFFFF

geetee24, 2017-10-20:
Any idea why a client.get returns about 50 FF hex chars and then the actual real data from the server?

Thanks.

mamama1, 2017-10-20:
Maybe you could share a minimal sketch with us which allows us to reproduce this.

Thanks

geetee24, 2017-10-20:
have you heard of this issue?  If so, is it a new issue?

I doubt that my shield is bad.   if i use the same code on a hard wired board,  to get the same file from the server, it works fine.

geetee24, 2017-10-20:
bool sendGET ( char* cszHost, char *cszPathFile, unsigned int *puiCharCountGot, bool bDumpToConsole, bool bDumpAsTextOrHex ) {

	ETHERNET_CLIENT client;

	int x=0; //for counting line feeds
	char lf=10; //line feed character

	*puiCharCountGot = 0;

	bool bRetSuccess = false;

	if ( client.connect ( cszHost, 80 ) ) {  
		// starts client connection, checks for connection

		if ( bDumpToConsole ) Serial.println ( "connected" );
	
		client.print ( "GET " );
		client.print ( cszPathFile );
		client.println ( " HTTP/1.1" );

		client.print ( "Host: " );
		client.println ( cszHost );

		client.println ( "Connection: close" );
		client.println();	//end of get request
	
		while ( client.connected() || client.available() ) {
			// data available

			char c = client.read();

			(*puiCharCountGot)++;

			if ( bDumpAsTextOrHex ) {
				// text

				if ( bDumpToConsole ) Serial.print ( c );

			} else {
				// hex

				if ( bDumpToConsole ) SerialPrintOutputDebugMessages ( "%x", c );
			}
    	}

		if ( *puiCharCountGot > 0 ) {

			bRetSuccess = true;

		} else {
			
			bRetSuccess = false;
		};
	} else {
		// no client connect
	
		if ( bDumpToConsole ) {

			Serial.println ( "connection failed" ); 
			Serial.println();
		};

		bRetSuccess = false;

	}

	client.stop(); //stop client

	return ( bRetSuccess );
}

--------------------

// gets file fron internet/ if success returns true else false if cant
bool sendGET ( char* cszHost, char *cszPathFile, unsigned int *puiCharCountGot, bool bDumpToConsole, bool bDumpAsTextOrHex ) {

	ETHERNET_CLIENT client;

	int x=0; //for counting line feeds
	char lf=10; //line feed character

	*puiCharCountGot = 0;

	bool bRetSuccess = false;

	if ( client.connect ( cszHost, 80 ) ) {  
		// starts client connection, checks for connection

		if ( bDumpToConsole ) Serial.println ( "connected" );
	
		client.print ( "GET " );
		client.print ( cszPathFile );
		client.println ( " HTTP/1.1" );

		client.print ( "Host: " );
		client.println ( cszHost );

		client.println ( "Connection: close" );
		client.println();	//end of get request
	
		while ( client.connected() || client.available() ) {
			// data available

			char c = client.read();

			(*puiCharCountGot)++;

			if ( bDumpAsTextOrHex ) {
				// text

				if ( bDumpToConsole ) Serial.print ( c );

			} else {
				// hex

				if ( bDumpToConsole ) SerialPrintOutputDebugMessages ( "%x", c );
			}
    	}

		if ( *puiCharCountGot > 0 ) {

			bRetSuccess = true;

		} else {
			
			bRetSuccess = false;
		};
	} else {
		// no client connect
	
		if ( bDumpToConsole ) {

			Serial.println ( "connection failed" ); 
			Serial.println();
		};

		bRetSuccess = false;
	}

	client.stop(); //stop client

	return ( bRetSuccess );
}

--------------------------

output is:

Opening port
Port open
Allocated 512 bytes for HTTP buffer
Using WIFI 101 shield

WiFi shield present
Firmware version: 19.4.4
Firmware version required : 19.5.2
MAC: F8:F0:5:F0:96:8E
Attempting to connect to SSID: xxxxxx
Encryption Type: WPA

Getting file from internet... connected
??????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????HTTP/1.1 200 OK
Date: Fri, 20 Oct 2017 05:48:44 GMT
Server: Apache/2.2
Last-Modified: Thu, 19 Oct 2017 18:40:57 GMT
ETag: "147ae60-4c-55beab2652040"
Accept-Ranges: bytes
Content-Length: 76
Connection: close
Content-Type: text/plain

Do not delete this file.
success. Read 473 characters

where the ???? are hex FF chars - garbage ...

does this help

geetee24, 2017-10-20:
any answer?

also. can you recommend arduino shields that work with your library?

sandeepmistry, 2018-01-02:
Hi @geetee24,

```arduino
	while ( client.connected() || client.available() ) {
		// data available

		char c = client.read();
```

This section of the sketch is incorrect. It is performing a read if the client is connected OR there is data for reading. Being connected doesn't necessarily mean there is data available for reading. `client.read()` will return `-1` in this case which maps to the 0xff you are seeing.

closed 2018-01-02 by sandeepmistry