arduino-libraries/WiFi101 #199: wpa2 enterprise

lorenzoromagnoli, 2017-10-31:
Can MKR1000 connect to wpa2 enterprise network like eduroam?

I remember I could do it with the Arduino YUN
http://forum.arduino.cc/index.php?topic=197267.0

sandeepmistry, 2017-10-31:
@lorenzoromagnoli it looks like it's possible from the ASF code: http://asf.atmel.com/docs/latest/common.components.wifi.winc1500.security_enterprise_network_example.sam4s_xplained_pro/html/main_8c.html

We don't have any API's exposed via the WiFi101 library and have not tested it ...

mastrolinux, 2019-07-18:
do we have any update on this one? Enterprise customers are starting asking for it in our support channel.

adriens, 2019-07-18:
Yes, pleeeeaaaazzzz 🙏 😺

adriens, 2019-07-18:
Thank you @lorenzoromagnoli , I'll give a try on this on my Yun rev2.

closed 2020-08-26 by ubidefeo