arduino-libraries/WiFi101 #201: Can't use functions from m2m_crypto.c

BrightSoul, 2017-11-20:
Hi there, I'm struggling to use the crypto chip directly on an Arduino MKR1000. In my case, I need to perform modular exponentiation to decrypt an RSA encrypted content.

Here's the BigInt_ModExp function I'm trying to invoke.
https://github.com/arduino-libraries/WiFi101/blob/d13219d61862f7abf847a9cc0c7e8d9b8101ac5c/src/driver/source/m2m_crypto.c#L418

What I did:
- Forked your repo;
- Added a BigInt_ModExp definition to m2m_crypto.h, since it was not defined there;
- Created a CryptoChip class as a wrapper to the BigInt_ModExp and other functions found in m2m_crypto.h;
- Used the CryptoChip class and for the love of me I couldn't get it to work.

Here's my code:
https://github.com/BrightSoul/WiFi101/blob/76edfcb0d867678fe175964682a29e50d02cdd57/src/CryptoChip.cpp#L16

And here's the error I'm getting when I compile the [Arduino Sketch](https://github.com/BrightSoul/WiFi101/blob/master/examples/CryptoChip/CryptoChip.ino).
```
CryptoChip.cpp:16: undefined reference to `BigInt_ModExp'
collect2.exe: error: ld returned 1 exit status
```
I am aware of [name mangling](https://en.wikipedia.org/wiki/Name_mangling#Name_mangling_in_C.2B.2B), but everything seems to be correct to me.
It's hard to understand how what's going on since m2m_crypto functions are not even referenced by any example or c++ class.

Thank for any help you might provide.

sandeepmistry, 2017-12-04:
Hi @BrightSoul,

You'll need to add a define to the library. Add: `#define CONF_CRYPTO_HW` here https://github.com/arduino-libraries/WiFi101/blob/d13219d61862f7abf847a9cc0c7e8d9b8101ac5c/src/bsp/include/nm_bsp_internal.h#L56

Please note, these functions don't use the crypto chip, rather the WINC1500 itself.

cc/ @agdl

closed 2017-12-04 by sandeepmistry

BrightSoul, 2017-12-04:
Hi @sandeepmistry, thanks for your response, I can now compile my sketch,
However, the BigInt_ModExp does not seem to return a correct response. Check my code here.
https://github.com/BrightSoul/WiFi101/blob/d9f1ce860ce237c6a37801b6b53245a9643a982b/examples/CryptoChip/CryptoChip.ino#L20

I'm passing these arguments to the BigInt_ModExp function:  base=3, exponent=2, modulo=7. The result should of course be 2 but I see this erratic behavior in the output.
```
Result is: 0x1B
Result is: 0x1B
Result is: 0x1B
Result is: 0x04
Result is: 0x1B
Result is: 0x1B
Result is: 0x04
Result is: 0x1B
```
Also, when I try with slightly bigger numbers as in [this sketch](https://github.com/BrightSoul/WiFi101/blob/c7677f6359bbe34df3a4afad22bb2517381099bd/examples/CryptoChip2/CryptoChip2.ino#L20), the result buffer is zeroed out.
```
Result is: 0x00 0x00 0x00 0x00
Result is: 0x00 0x00 0x00 0x00
Result is: 0x00 0x00 0x00 0x00
Result is: 0x00 0x00 0x00 0x00
```
Tried many times, with different numbers but I couldn't get a correct result.
Should I open a new issue?

sandeepmistry, 2017-12-04:
Hi @BrightSoul, great to hear that suggestion worked for compiling the sketch.

I would suggest you contact Microchip/Atmel support about this, as those source files come from the ASF SDK: http://www.microchip.com/support/hottopics.aspx

BrightSoul, 2017-12-04:
Will do, thank you. 👍

BrightSoul, 2018-01-21:
Hi @sandeepmistry,
a few weeks passed and I managed to get some help from the Microchip/Atmel support. They provided a sample project for their Atmel Studio IDE which I then ported to an Arduino sketch. Well, it's still not working. Not the modular exponentiation, not the sha256 calculation and not the rsa signature. 
I'm think I'm not "initializing the hardware" correctly, so I'll do some more experiments in the next few days.

In the meantime, I'd like to ask you: do you plan on releasing some other library other than WiFi101 for people who just need to use cryptographic functions? A Crypto101, or something like this.
Thank you, it would be much appreciated. This problem is driving me insane :D