arduino-libraries/WiFi101 #208: Wifi Library hangs during UDP processing

klausj1, 2018-01-08:
(See #204 / used Library version from #204 before merging.

Yesterday my sketch blocked after ~20 days. I do not have a serial log, but from the LED state I can see that it hangs in reading the time via UDP (code from this function see below, it hangs between `LEDsetAction(timeLED, HIGH);` and `LEDsetAction(timeLED, LOW);`, because the timeLED was on).

I assume it hangs when WLAN is turned off during or shortly before calling this function. The blue WLAN LED on the WLAN shield was on, so somehow the disconnect was not realized.

Maybe you have the chance to find something. Here is the code (again, I disconnect the WiFi if I do not get an answer, trying a "self-repair"):

```
time_t getNtpTime()
{
    Serial.println("Fetching current time from internet... ");
    if (WiFi.status() != WL_CONNECTED) {
        Serial.println("No WiFi connection, aborting.");
        return 0;
    }
    LEDsetAction(timeLED, HIGH);

    Serial.print("WLAN State: ");
    Serial.println(WiFi.status());

    Udp.begin(localPort);
    delay(100);

    IPAddress ntpServerIP; // NTP server's ip address
    int tries = 0;
    int bytesInBuffer = 0;

    unsigned long startTime;
    unsigned long udpTimeOut = 1500;

    while (tries < 5)
    {
        if (WiFi.status() != WL_CONNECTED)
        {
            LEDsetAction(timeLED, LOW);
            Udp.stop();
            return 0;
        }

        // discard any previously received packets 
        unsigned long startTime = millis();

        bytesInBuffer = Udp.parsePacket();
        wdt_reset();
        while (bytesInBuffer > 0 && ((millis() - startTime) < udpTimeOut))
        {
            Serial.print("reading previously received packets ... Count: ");
            Serial.print(bytesInBuffer);
            if (bytesInBuffer > NTP_PACKET_SIZE) bytesInBuffer = NTP_PACKET_SIZE;
            Udp.read(packetBuffer, NTP_PACKET_SIZE);
            bytesInBuffer = Udp.parsePacket();
            delay(10);
        }
        if ((millis() - startTime) >= udpTimeOut)
            Serial.println("Could not finish reading previously received packets.");

        if (WiFi.status() != WL_CONNECTED)
        {
            LEDsetAction(timeLED, LOW);
            Udp.stop();
            return 0;
        }

        Serial.println("Transmit NTP Request");
        // get a random server from the pool
        WiFi.hostByName(ntpServerName, ntpServerIP);
        Serial.print(ntpServerName);
        Serial.print(": ");
        Serial.println(ntpServerIP);
        sendNTPpacket(ntpServerIP);
        startTime = millis();
        while (millis() - startTime < udpTimeOut) {
            int size = Udp.parsePacket();
            //Serial.print("NTP size: ");
            //Serial.println(size);
            if (size >= NTP_PACKET_SIZE) {
                Serial.println("Receive NTP Response");
                Udp.read(packetBuffer, NTP_PACKET_SIZE);  // read packet into the buffer
                unsigned long secsSince1900;
                // convert four bytes starting at location 40 to a long integer
                secsSince1900 = (unsigned long)packetBuffer[40] << 24;
                secsSince1900 |= (unsigned long)packetBuffer[41] << 16;
                secsSince1900 |= (unsigned long)packetBuffer[42] << 8;
                secsSince1900 |= (unsigned long)packetBuffer[43];
                if (secsSince1900 == 0)
                {
                    Serial.print("Wrong time (zero) received, ignoring time.");
                }
                else
                {
                    time_t t = secsSince1900 - 2208988800UL; // 2208988800UL: difference between 1900 (NTP) and 1970 (unix time)
                    Serial.print("Currenttime: ");
                    Serial.println(currentTime);
                    if ((currentTime < 86400*365) || (abs(t - currentTime) < 86400))
                    {
                        Serial.println("Setting time.");
                        LEDsetAction(timeLED, LOW);
                        Udp.stop();
                        return t;
                    }
                    if (abs(t - currentTime) >= 86400)
                    {
                        Serial.print("Wrong time received, time difference to current time is too high. Ignoring time.");
                    }
                    Serial.print("NTP: Should not be here!!!");
                }
                delay(100);
            }
            else delay(10);
        }
        tries++;
    }
    wdt_reset();
    Serial.println("No valid NTP Response after 5 times :-(");
    Serial.print("WLAN State: ");
    Serial.println(WiFi.status());
    Udp.stop();

    WlanDisconnect();

    LEDsetAction(timeLED, LOW);
    return 0; // return 0 if unable to get the time
}
```

klausj1, 2018-01-27:
Since some days again my sketch stopped working. This time it was somehow different, but still related to UDP: The sketch did not block, but in the same code as above no answer was received from any UDP-Call. The log was as follows:

```
Fetching current time from internet... 
WLAN State: 3
Transmit NTP Request
us.pool.ntp.org: 208.88.126.235
Transmit NTP Request
us.pool.ntp.org: 208.88.126.235
Transmit NTP Request
us.pool.ntp.org: 208.88.126.235
Transmit NTP Request
us.pool.ntp.org: 208.88.126.235
Transmit NTP Request
us.pool.ntp.org: 208.88.126.235
No valid NTP Response after 5 times :-(
WLAN State: 3
Disconnecting WiFi ...
Disconnecting done.
WLAN State: 6
```
And this happened in an endless loop (done by my sketch not shown in the code above).

Maybe some fixes done for TCP were not done for UDP?

Klaus

sandeepmistry, 2018-02-05:
@klausj1 do you have a sketch and method to reproduce this without waiting 20 days?

UDP writes call the ASF's `sendto` API directly: https://github.com/arduino-libraries/WiFi101/blob/master/src/utility/WiFiSocket.cpp#L303-L312

We could try adding a check to `hif_receive_blocked` to bail early though. However, best to narrow it down to something that can be reproduced first.

```
if (hif_receive_blocked) {
 // ...
}
```

klausj1, 2018-03-11:
I tried to reproduce this with disconnecting the WLAN (via code and/or via turning it off) while UDP-data are pending, but I did not succeed, it worked as expected.

The only thing I found is when you call udp.begin twice, then Udp.parsePacket() will not return true anymore, until you call a WiFi.disconnect().

It should be easy to reproduce, however, below is the sketch.

Klaus

```
/*

 Udp NTP Client

 Get the time from a Network Time Protocol (NTP) time server
 Demonstrates use of UDP sendPacket and ReceivePacket
 For more on NTP time servers and the messages needed to communicate with them,
 see http://en.wikipedia.org/wiki/Network_Time_Protocol

 created 4 Sep 2010
 by Michael Margolis
 modified 9 Apr 2012
 by Tom Igoe

 This code is in the public domain.

 */

#include <SPI.h>
#include <WiFi101.h>
#include <WiFiUdp.h>

int status = WL_IDLE_STATUS;
#include "arduino_secrets.h" 
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;            // your network key Index number (needed only for WEP)

unsigned int localPort = 2390;      // local port to listen for UDP packets

IPAddress timeServer(129, 6, 15, 28); // time.nist.gov NTP server

const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message

byte packetBuffer[ NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets

// A UDP instance to let us send and receive packets over UDP
WiFiUDP Udp;

bool firstRun = true;

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to WiFi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
[... 135 lines cut ...]
    Serial.println(epoch);

    // print the hour, minute and second:
    Serial.print("The UTC time is ");       // UTC is the time at Greenwich Meridian (GMT)
    Serial.print((epoch  % 86400L) / 3600); // print the hour (86400 equals secs per day)
    Serial.print(':');
    if ( ((epoch % 3600) / 60) < 10 ) {
      // In the first 10 minutes of each hour, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.print((epoch  % 3600) / 60); // print the minute (3600 equals secs per minute)
    Serial.print(':');
    if ( (epoch % 60) < 10 ) {
      // In the first 10 seconds of each minute, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.println(epoch % 60); // print the second
    firstRun = false;
  }
  else Serial.println("No UDP data received");
  // wait ten seconds before asking for the time again
  delay(10000); 
  return;
}
```

sandeepmistry, 2018-04-09:
@klausj1 can this be closed now?

klausj1, 2018-04-09:
As I cannot really reproduce it, I close it. If something pops up, I will create a new issue.

closed 2018-04-09 by klausj1