arduino-libraries/WiFi101 #209: No WiFi101 example sketches work on Arduino Due.

SamuraiJones, 2018-01-09:
Hello,

Purchased an Adafruit ATWINC1500 breakout board.  Has never worked on Arduino Due.  Borrowed an Uno from a mate and tested, the ATWINC1500 does work on the Uno so the WiFi board is not the issue.  

A couple of issues. The first being the Arduino IDE 1.8.5 does allow me to update the firmware on the ATWINC1500, but I am only shown the 19.5.2 version as the latest version.  I poked around on github and found the latest version of the updater plugin (https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin) and swapped out the .bin file in the 19.5.2 folder of my IDE install for the .bin file that was in the 19.5.4 folder in the plugin.  End result was the firmware on the ATWINC1500 was updated to 19.5.4 and when connected to the Uno the CheckWiFi101FirmwareVersion sketch showed it to be 19.5.4.

Now when wired up to the Due, I am using the 6 pin SPI header for the SCK, MOSI, MISO and other digital pins for the CS (52), INT(50), RST(51), and wired 5V to the EN (Enable) pin on the ATWINC1500 breakout board.  I have used `WiFi.setPins(52,50,51);` as the first line of the `void setup() {` section of the sketch.  I have used an RA8875 breakout board on the 6 pin SPI header for the SCK, MOSI, MISO and it works as expected, so the SPI works on the Due.  I dont have the RA8875 connected when trying the ATWINC1500 (I am aware of the issue with using the RA8875 with more than one device on the SPI bus).  

Running the CheckWiFi101FirmwareVersion example in the serial monitor simply displays:

WiFi101 firmware check.

WiFi101 shield:

In the sketch it hangs when it tries to execute: 

`if (WiFi.status() == WL_NO_SHIELD) {`

Now I'm not a developer but poking around in the WiFi.cpp file, if I comment out 	`m2m_wifi_handle_events(NULL);` lines the example sketches seem to run - sort of.  With `m2m_wifi_handle_events(NULL);` commented out in the `WiFiClass::status()` function it returns the firmware version as expected.  If I comment out all occurrences of `m2m_wifi_handle_events(NULL);` in WiFi.cpp all the example sketches seem to run but with varying results.  For example the ScanNetworks returns the following in the Serial Monitor:

MAC: F8:F0:5:F7:54:C
Scanning available networks...
** Scan Networks **
number of available networks:0

This then repeats (minus the MAC Address) every 10 seconds as expected, but does not find any networks.  When connected to the Uno it did find available networks as I was expecting. 

Anyway, hopefully this can be fixed soon, and hopefully my laymen investigation and notes about commenting out the `m2m_wifi_handle_events(NULL);` lines in WiFi.cpp help in investigating the root cause. 

Many thanks.

JasonBCC, 2018-01-23:
Hi SamuraiJones,
I have the same issue, using custom hardware PCB with an onboard SAM3X8E (aka arduino due) and a winc1500 rev B module (aka wifi 101 shield) been trying to use the new upgraded WiFi101 library to eradicate some of the bugs and it just hangs thought i was going mad until I found your post, same issues as you. I have been using successfully the (https://github.com/adafruit/Adafruit_WINC1500) library  a fork of an older version of the WiFi101 library that added support for different pinouts.  So whatever the changes to the Wifi101 library after this fork have been made has made the library incompatible with the Arduino Due.  I have not delved too deep under under the bonnet (/hood!) as it quickly becomes difficult to follow but believe it may have something to do with some of the processor configuration i.e. if is an arduino it must then be and avr chip, else it is a SMAD?? and then the subsequent loading of the requisite libraries at compile time?? hence it does not work with the DUE??  could be way off line but would be a good starting point for someone with a much more intimate knowledge of the library than me...

I wanted to upgrade as had issues if i opened an UDP port after a SSLClient it hangs however if I open the UDP port first and connect to a time server then the SSLClient, all is well, story for another post and day...

Thanks,

Jason

sandeepmistry, 2018-02-05:
Using a Due + WiFi101 shield, I am able to reproduce this.

We are hitting a "false" interrupt path in the ASF code and looping indefinitely: https://github.com/arduino-libraries/WiFi101/blob/8933d9063cb210291ab78ca5dfa97a06e05c3f83/src/driver/source/m2m_hif.c#L602

sandeepmistry, 2018-02-05:
@SamuraiJones @JasonBCC could you please try the changes from PR https://github.com/arduino-libraries/WiFi101/pull/216 and provide your feedback.

closed 2018-02-14 by sandeepmistry

sandeepmistry, 2018-02-14:
The new 0.15.1 library release includes a fix for this and will be available in the library manager shortly. Please let us know if you have further issues.

PR #216 (2018-02-14): Fix Due not working
This fixes #209.

@facchinm the Due seems to be getting "false" interrupts for some reason now. I made https://github.com/arduino-libraries/WiFi101/commit/48caad4021c2d405b143afb0221b96ee69d43016 to ignore these in this PR, however there might be a better way. The "false interrupt" condition basically results in an infinite loops anyways, so I think we can safely ignore it.

I also noticed not having the shield pulled in would result in `WiFi.status()` hanging because it tries to set the error LED on the shield. https://github.com/arduino-libraries/WiFi101/commit/ba2b2a9fa6 [... cut]