arduino-libraries/WiFi101 #210: Implementing WiFi Direct (P2P Mode)

dionikink, 2018-01-10:
Someone requested WiFi Direct to be implemented a while ago, but back then the issue was closed because Access Points were just added.

I would like to resuggest implementing a P2P Mode. I am currently working on an implementation, but my progress has been stalled. The ATWINC1500 chip doesn't seem to enter P2P listening mode, although I'm not 100% certain that's the issue.

You can find my (currently not working) implementation [here](https://github.com/dionikink/Wifi101P2P). My need for this mode is quite urgent and I'm definitely willing to finish writing a complete implementation. However, I could really use a fresh view because I have no clue what is going wrong currently. The actual implementation can be found in [this method](https://github.com/dionikink/Wifi101P2P/blob/master/src/WiFi.cpp#L459).

Thanks in advance!

sandeepmistry, 2018-01-10:
See the WINC1500 firmware change log: http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel_WINC1500_Software_Release_Notes.pdf

> 2. Changes since the last release (version 19.5.2)
> - WLAN Features:
>   - Removed P2P (Wi-Fi direct) client mode feature.

I don't think this feature should be implemented in this library, as the firmware side was removed.

dionikink, 2018-01-10:
I see... that's unfortunate. In that case I agree with not implementing it. 

One question, though (for personal use): would the P2P Mode work if I were to downgrade the driver version back to 19.5.2, before it was removed? The fact that it has been removed explains why my code didn't work, so I'm thinking that a downgrade might fix that.

sandeepmistry, 2018-01-10:
If you downgrade to 19.4.4 is should work, I've only tested with an Android device as per the https://hackaday.io/project/11734-1sheeld-in-0/log/38145-implementing-wifi-direct-in-arduino-library-wifi101 link mentioned in #92.

closed 2018-01-10 by sandeepmistry