arduino-libraries/WiFi101 #213: Problems with the WeMos D1

AmmannFabian, 2018-01-23:
#include <PubSubClient.h>
#include <SPI.h>
#include <WiFi101.h>

/*
#define DHTPIN 2 // DHT Sensor connected to digital pin 2.
#define DHTTYPE DHT11 // Type of DHT sensor.
#define LIGHTPIN A0 // Analog light sensor connected to analog pin A0.
*/

char ssid[] = "xxx"; //  Change this to your network SSID (name).
char pass[] = "xxx";  // Change this your network password
char mqttUserName[] = "xxx";  // Can be any name.
char mqttPass[] = "xxx";  // Change this your MQTT API Key from Account > MyProfile.
char writeAPIKey[] = "xxx";    // Change to your channel Write API Key.
long channelID = xxx;

static const char alphanum[] ="0123456789"
                              "ABCDEFGHIJKLMNOPQRSTUVWXYZ"
                              "abcdefghijklmnopqrstuvwxyz";  // For random generation of client ID.

WiFiClient client;                         // Initialize the Wifi client library.
// DHT dht(DHTPIN, DHTTYPE); // Initialize DHT sensor.

PubSubClient mqttClient(client); // Initialize the PuBSubClient library.
const char* server = "mqtt.thingspeak.com"; 

unsigned long lastConnectionTime = 0; 
const unsigned long postingInterval = 20L * 1000L; // Post data every 20 seconds.

void setup() 
{
  Serial.begin(9600);
  int status = WL_IDLE_STATUS;  // Set a temporary WiFi status.

  // Attempt to connect to WiFi network
  while (status != WL_CONNECTED) 
  {
    status = WiFi.begin(ssid, pass); // Connect to WPA/WPA2 Wi-Fi network.
    delay(5000);
  }

  Serial.println("Connected to wifi");
  mqttClient.setServer(server, 1883);   // Set the MQTT broker details.
}

void loop() 
{
 // Reconnect if MQTT client is not connected.
  if (!mqttClient.connected()) 
  {
    reconnect();
  }

  mqttClient.loop();   // Call the loop continuously to establish connection to the server.

  // If interval time has passed since the last connection, Publish data to ThingSpeak
  if (millis() - lastConnectionTime > postingInterval) 
  {
    mqttpublish();
  }
}

void reconnect() 
{
  char clientID[10];

  // Loop until we're reconnected
  while (!mqttClient.connected()) 
  {
    Serial.print("Attempting MQTT connection...");
    // Generate ClientID
    for (int i = 0; i < 8; i++) {
        clientID[i] = alphanum[random(51)];
    }

    // Connect to the MQTT broker
    if (mqttClient.connect(clientID,mqttUserName,mqttPass)) 
    {
      Serial.println("connected");
    } else 
    {
      Serial.print("failed, rc=");
      // Print to know why the connection failed.
      // See http://pubsubclient.knolleary.net/api.html#state for the failure code explanation.
      Serial.print(mqttClient.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}

// Use this function instead to publish to a single field directly.  Don't forget to comment out the above version.

void mqttpublish() {
  
  float t = 5; // Read temperature from DHT sensor.

  String data = String(t, DEC);
  int length = data.length();
  char msgBuffer[length];
  data.toCharArray(msgBuffer,length+1);
  Serial.println(msgBuffer);
  
  // Create a topic string and publish data to ThingSpeak channel feed. 
  String topicString ="channels/" + String( channelID ) + "/publish/"+String(writeAPIKey);
  length=topicString.length();
  char topicBuffer[length];
  topicString.toCharArray(topicBuffer,length+1);
 
  mqttClient.publish( topicBuffer, msgBuffer );

  lastConnectionTime = millis();
}

I got the following fault:

Arduino: 1.8.1 (Windows 7), Board: "WeMos D1 R1, 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 921600"

C:\Users\Anwender\Documents\Arduino\libraries\WiFi101\src\WiFiMDNSResponder.cpp:24:26: fatal error: avr/pgmspace.h: No such file or directory

 #include <avr/pgmspace.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

The library avr/pgmspace.h had not been included in the sketch by me. I tested the sketch with this library included too but that also didn´t work.

Had anybody the same problem and knows how to solve it?

ogiewon, 2018-01-23:
The Wemos D1 is based on the ESP8266 chip with built in WiFi. You don’t need the WiFi101 library to use it. Just use the ESP8266 Arduino IDE integration.

sandeepmistry, 2018-02-05:
@AmmannFabian are you trying to connect a WINC1500 board to the ESP8266?

Or just you the builtin WiFi, if so please see @ogiewon's comment above.

I'm going to close this for now, but please re-open if you have a ESP8266 connected to a WINC1500 board.

closed 2018-02-05 by sandeepmistry