arduino-libraries/WiFi101 #214: Lockup upon call for client after WiFi.end()

BrunDog, 2018-01-27:
Hi,

I have found an odd aberration that afflicts certain WINC1500 modules. For example, the code below will cause a lockup on a ATWINC1500-MR210xB D/C: M1613 but not on D/C: D1532 or D1621. These are all Adafruit Feather M0 WINC1500 modules.

The lockup occurs on the call of 'WiFiClient client = server.available();' following the call to WiFi.end();. Certainly, the call to check for a client may not seem normal when disconnected (it is in another application I am using), but shouldn't cause a hard lockup either, so I thought I would post it.

`#include <SPI.h>
#include <WiFi101.h>

#include "arduino_secrets.h"
char ssid[] = SECRET_SSID;
char pass[] = SECRET_PASS;
IPAddress IP = {192, 168, 1, 240};
int status = WL_IDLE_STATUS;
int restartLevel = 0;

WiFiServer server(23);

boolean alreadyConnected = false;
unsigned long oldtime;

void setup() {
  WiFi.setPins(8, 7, 4, 2);
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  while (!Serial) {
    ;
  }
  initNetwork();
}

void loop() {
  WiFiClient client = server.available();

  if (client) {
    if (!alreadyConnected) {
      client.flush();
      Serial.println("We have a new client");
      client.println("Hello, client!");
      alreadyConnected = true;
    }

    if (client.available() > 0) {
      char thisChar = client.read();
      server.write(thisChar);
      Serial.write(thisChar);
      if (thisChar == '!') restartLevel = 1;
    }
  }
  if ((millis() - oldtime) >= 3000) {
    printWiFiStatus();
    if (digitalRead(LED_BUILTIN)) digitalWrite(LED_BUILTIN, LOW);
    else digitalWrite(LED_BUILTIN, HIGH);
    if (restartLevel) restartSequence();
    oldtime = millis();
  }
}

void printWiFiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());
  IPAddress ipx = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ipx);
  long rssi = WiFi.RSSI();
  Serial.print("RSSI:");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void initNetwork() {
  WiFi.setPins(8, 7, 4, 2);
  WiFi.config(IP);
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("No WiFi hardware");
    while (true);

  status = WiFi.begin(ssid, pass);
  while ( status != WL_CONNECTED) {
    Serial.print("Connecting: ");
    Serial.println(ssid);

    status = WiFi.begin(ssid, pass);
    delay(10000);
  }
  server.begin();
  printWiFiStatus();
}

void restartSequence() {
  switch (restartLevel) {
    case 1:
      WiFi.disconnect();
      restartLevel = 2;
      Serial.println("WiFi Disconnected");
      break;
    case 2:
      WiFi.end();
      restartLevel = 3;
      Serial.println("WiFi Ended");
      break;
    case 3:
      initNetwork();
      restartLevel = 0;
      break;
  }
}`

sandeepmistry, 2018-02-05:
Hi @BrunDog,

Thanks for letting us know about this, I was able to reproduce it on a MKR1000. (Note your sketch did not compile successfully so I had to make some tweaks).

Could you please try the changes I proposed in pull request #215 and provide your feedback. Thanks.

BrunDog, 2018-02-05:
Sorry... it looks like I somehow deleted a closing brace on the copy over.

Excuse the dummy... how do I easily test this... just copy over the affected library files?

sandeepmistry, 2018-02-05:
@BrunDog no worries, and thanks for asking :)

You can download a zip of the modded lib [here](https://github.com/sandeepmistry/WiFi101/archive/issue-214-fixes.zip). Then backup your `<SKETCHBOOK>/libraries/WiFi101` folder, and replace it with the files in the diff. The `<SKETCHBOOK>` directory can be found in the IDE's preferences.

BrunDog, 2018-02-06:
Sorry... for some odd reason it will not compile now... If I swap the libraries it compiles ( I am removing the old library completely from the sketchbook file and replacing it with this one). I don't have time tonight to debug this. I do see setPins in the header, keywords, and .cpp files, so not sure what is going on just yet.

C:\Users\pbrun\Documents\Arduino\WiFi101Test\WiFi101Test.ino: In function 'void setup()':

WiFi101Test:19: error: 'class WiFiClass' has no member named 'setPins'

   WiFi.setPins(8, 7, 4, 2);

        ^

C:\Users\pbrun\Documents\Arduino\WiFi101Test\WiFi101Test.ino: In function 'void initNetwork()':

WiFi101Test:70: error: 'class WiFiClass' has no member named 'setPins'

   WiFi.setPins(8, 7, 4, 2);

        ^

C:\Users\pbrun\Documents\Arduino\WiFi101Test\WiFi101Test.ino: In function 'void restartSequence()':

WiFi101Test:93: error: 'class WiFiClass' has no member named 'end'

       WiFi.end();

            ^

exit status 1
'class WiFiClass' has no member named 'setPins'

sandeepmistry, 2018-02-06:
@BrunDog could you please enable verbose compile output in the IDE preferences and provide the full output. What directory is the patched library in now?

BrunDog, 2018-02-06:
Ignore me... somehow I changed the library reference to WiFi.h instead of WiFi101.h. The verbose compilation mode pointed it out, and then I realized somehow I changed the definition above (I think it was because I was testing it with ESP32 hardware).

Anyway, yessir... that seemed to fix the problem!
I also have the WiFi 101 and Due problem noted in #209. I look forward to these two getting fixed. Thank you for your help and responsiveness with these! I know its thankless tiresome work sometimes!

closed 2018-02-14 by sandeepmistry

sandeepmistry, 2018-02-14:
@BrunDog thank you for your feedback a new 0.15.1 library release which includes both fixes will be available in the library manager shortly.

PR #215 (2018-02-14): Fixes for issue #214
Fixes #214.

The following sketch based on @BrunDog's sketch can be used to reproduce the issue on a MKR1000:

```arduino
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "SSID";
char pass[] = "PASS";
//IPAddress IP = {192, 168, 1, 240};
int status = WL_IDLE_STATUS;
int restartLevel = 0;

WiFiServer server(23);

boolean alreadyConnected = false;
unsigned long oldtime;

void setup() {
  //  WiFi.setPins(8, 7, 4, 2);
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  while (!Serial) {
    ;
  }
  initNetwork();
}

void loop() {
  WiFiClient client = server.availab [... cut]