arduino-libraries/WiFi101 #217: No certificate-based auth e.g. WiFi.setClientCertificate()

TmpR, 2018-02-06:
There's a lot of cross-promotion between Arduino and Amazon, specifically, the MKR1000 is aimed at IoT and Amazon offer AWS IoT. I see the two frequently advertised together, although Amazon only offer an SDK for the Yun. The MKR1000 is much better suited to IoT than the Yun.

ASW IoT only supports cert-based authentication but this is not supported by WiFi101. This means the MRK1000 won't work with AWS IoT out of the box.

Is there any way this functionality could be added? 

There is an example here using the old WiFi lib  (Arduino WiFi shield):  https://github.com/Ameba8195/Arduino/blob/master/hardware_v2/libraries/MQTTClient/examples/amazon_awsiot_basic/amazon_awsiot_basic.ino

sandeepmistry, 2018-02-14:
Hi @TmpR,

Unfortunately the current WINC1500 firmware doesn't support dynamic configuration of cert based authentication. I would suggest you contact Microchip/Atmel support to request this feature: http://www.microchip.com/support/hottopics.aspx

Please let us know of the support ticket # if you decide to proceed with this.

TmpR, 2018-02-14:
Thanks for getting back to me. So the MRK1000 won't work with ASW IoT until Amtel add this feature? Bummer, I've entered a competition promoted by Amazon and Arduino to build an Alexa powered IoT device based on Arduino hardware.  The MRK1000 would have been perfect as the Yún is too big. :(

sandeepmistry, 2018-02-15:
To do it in the way you proposed, yes.

As of WINC1500 firmware 19.5.x and later, you can store a RSA public cert and private key on it's flash. However, you will have to use Microchips/Atmel's Window's only tools to get them on the WINC1500 at this time.

ShridharK91, 2018-03-02:
So if I store the root cert, private key and client certificate on WINC1500 flash using Atmel's windows tool, would I be able to connect to AWS IOT using wifiSSLclient functions in arduino wifi library?

gjt211, 2019-04-19:
Hi @sandeepmistry and others participating in this thread.
I found a document from Atmel that seems to indicate that certificates can be downloaded and programmed into the WINC1500 module without using the Microchip/Atmel Window's only tools.

The following link is to a pdf that shows basics on how to download a certificate over https and add/replace the certificates already in the WINC1500 using example code and Atmel Studio. This is code for the SAM/WINC1500 combo so should be able to work with the MKR1000.
[http://ww1.microchip.com/downloads/en/DeviceDoc/50002703A.pdf](url)

Is it possible that this can also be done with Arduino and WiFi101?

sandeepmistry, 2019-04-23:
> Is it possible that this can also be done with Arduino and WiFi101?

It's possible to do it, but not something we support in this library.

If you are interested in using a private key, I suggest to following the tutorial for AWS IoT Core found here: https://github.com/arduino/ArduinoCloudProviderExamples

They use the BearSSL TLS/SSL stack in combination with the boards ATECC508a crypto chip and will run on all MKR family boards that can support TCP connections.

gjt211, 2019-04-24:
Hi @sandeepmistry, thanks for your reply.

In my case, I don't use/need private keys. I had a good look through the link you provided thanks and it has provided me more things to think about and consider.

I am grateful, thankful, and highly appreciative of the no doubt countless hours/days/months/years of work by many dedicated and amazing people (such as yourself) that go into these libraries for the less skilled people like myself to use. 

In my scenario, web servers change, certificates change, certificate providers can change etc. In my infrastructure, I could tell a sensor that a new certificate is available and download it. I already do this with firmware as none of my sensors are physically accessible and can only be updated this way remotely.

My problem is my lack of skills. I know what I want to do, but I don't know how to do it. It has taken me a few years to get my sensors working the way I need, however I am now worried that my server will need to be replaced and as a result the certificates will change. My plan was to release a firmware update that will allow the sensors to obtain the new certificate if/when one becomes available.

Keep up the great work!

sandeepmistry, 2019-09-18:
Hi Everyone, we don't plan to support this feature, so I'm closing and marking as won't fix for now.

closed 2019-09-18 by sandeepmistry