arduino-libraries/WiFi101 #220: Error triyng to run WifiSSLClient example

mikfaina, 2018-02-12:
Hi all,
i'm trying to run the example here but  I get compilation error:

`Collected 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> v1.0
|-- <WiFi101> v0.15.0
|   |-- <SPI> v1.0
Compiling .pioenvs\lolin32\src\main.o
Archiving .pioenvs\lolin32\lib90c\libSPI.a
Compiling .pioenvs\lolin32\lib136\WiFi101_ID299\WiFi.o
Compiling .pioenvs\lolin32\lib136\WiFi101_ID299\WiFiMDNSResponder.o
Compiling .pioenvs\lolin32\lib136\WiFi101_ID299\WiFiSSLClient.o
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFiMDNSResponder.cpp:24:26: fatal error: avr/pgmspace.h: No such file or directory
compilation terminated.
*** [.pioenvs\lolin32\lib136\WiFi101_ID299\WiFiMDNSResponder.o] Error 1
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp: In member function 'int WiFiClass::init()':
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp:300:46: error: invalid conversion from 'void (*)(uint8_t*, uint32_t) {aka void (*)(unsigned char*, unsigned int)}' to 'tpfAppResolveCb {aka void (*)(unsigned char*, long unsigned int)}' [-fpermissive]
registerSocketCallback(socket_cb, resolve_cb);
^
In file included from C:\users\polizio\.platformio\lib\WiFi101_ID299\src\utility/WiFiSocket.h:24:0,
from C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp:36:
C:\users\polizio\.platformio\lib\WiFi101_ID299\src/socket/include/socket.h:1047:14: note:   initializing argument 2 of 'void registerSocketCallback(tpfAppSocketCb, tpfAppResolveCb)'
NMI_API void registerSocketCallback(tpfAppSocketCb socket_cb, tpfAppResolveCb resolve_cb);
^
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp: At global scope:
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp:1124:10: error: prototype for 'uint32_t WiFiClass::getTime()' does not match any in class 'WiFiClass'
uint32_t WiFiClass::getTime()
^
In file included from C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi.cpp:38:0:
C:\users\polizio\.platformio\lib\WiFi101_ID299\src\WiFi101.h:157:16: error: candidate is: long unsigned int WiFiClass::getTime()
unsigned long getTime();`

What's wrong?

sandeepmistry, 2018-02-14:
Hi @mikfaina,

What board are you trying to use the WiFi101 library with? Also, could you please try again but use the Arduino IDE to compile with verbose output enabled and provide the full output here. Thanks.

mikfaina, 2018-02-22:
I have Lolin32

sandeepmistry, 2018-07-23:
This WiFi101 library is not compatible with the ESP32.

closed 2018-07-23 by sandeepmistry