arduino-libraries/WiFi101 #221: MKR1000 with 15.1

s-simma, 2018-02-15:
With version 15.1 my web-server doesn't get any request from the browser.
Version 15.0 runs without any problems.

sandeepmistry, 2018-02-15:
Hi @s-simma,

Could you please provide a sketch to reproduce this issue?

The [SimpleWebServerWiFi](https://github.com/arduino-libraries/WiFi101/blob/master/examples/SimpleWebServerWiFi/SimpleWebServerWiFi.ino) and [WiFiWebServer](https://github.com/arduino-libraries/WiFi101/blob/master/examples/WiFiWebServer/WiFiWebServer.ino) example sketches are working for me.

s-simma, 2018-02-15:
Hi,
Thank's for your help. I'll try to make a small sketch to reproduce my problem.

Are you shure this code will work if you have more than 1 "WiFiServer" in use ?
---
---
	if (_socket != -1 && WiFiSocket.listening(_socket)) {
		WiFiSocket.close(_socket);
		_socket = -1;
	}
----
----

s-simma, 2018-02-18:
Hi,
It is very easy to simmulate the problem.

I have two server running (WEBserver, EEPServer).
With version 15.0 everything was ok. Both servers get a client if there is a connection request from remote.

With version 15.1 **only the last server** started "server.begin()" create a client.
e.g.
    // Save WEB-Page to EEProm
    EEPServer.begin();    
    // Start WEB Server
    WEBserver.begin();

WEBserver gets a client, EEPServer **does not** get a client

if i change the order of starting
    // Start WEB Server
    WEBserver.begin();
    // Save WEB-Page to EEProm
    EEPServer.begin();    

WEBserver **doe's not** get a client, EEPServer gets a client

Code for checking client available:
  // listen for incoming clients
  WiFiClient EEPClient = EEPServer.available();
  if (EEPClient) {  // got client?
....
....

closed 2018-02-20 by sandeepmistry

sandeepmistry, 2018-02-20:
Hi @s-simma,

There will be a new v0.15.2 release available in the library manager shortly, which includes a fix for this.

Thank you for letting us know about this.

PR #222 (2018-02-20): Initialize WiFiServer::_socket variable
Fixes #221.