arduino-libraries/WiFi101 #223: Can't create WPA PSK Access Point with password

Freezystem, 2018-02-24:
Hi, 
I'm using version `0.15.2` of the WiFi101 lib and an Arduino MKR1000 with firmware version `19.5.4`

Although I notice this line in the `WiFi101.h` :
```cpp
uint8_t beginAP(const char *ssid, const char* key);
``` 
I'm unable to create an authenticated Access Point using `WPA PSK`.

I'm simply doing:
```cpp
status = WiFi.beginAP("MKR1000", "secret");
```
And I'm always getting this status back `WL_AP_FAILED`.

I'm I doing it wrong ?

sandeepmistry, 2018-02-26:
Hi @Freezystem,

Can you please try again with a password that is 8 characters or longer, WPA passwords have a minimum length of 8 characters.

Freezystem, 2018-02-26:
Oh my bad.. It's working great with an 8 characters password. 

Thank you a lot for your time,
Regards.

closed 2018-02-26 by Freezystem