arduino-libraries/WiFi101 #227: Can't connect to wifi network

hiteule, 2018-03-23:
Hi,

Here is my setup : 
- Arduino MKR1000 with a WINC1501 Model B (I have this problem with a new board I just bought this week and another buyed month ago)
- WINC1501 firmware 19.5.4
- Arduino IDE 1.8.3
- WiFi101 0.15.2

I have many wifi network at office : 

       maxou:~$ airport -s
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                           Opera e4:aa:5d:be:17:7c -91  64      Y  FR NONE
                DIRECT-aL-BRAVIA aa:a7:95:11:ca:59 -75  11      Y  -- WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:04:c3:49:a5 -64  11      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:04:c4:d2:a5 -79  6       Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 9e:15:44:50:59:6e -66  6       Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 9e:15:44:50:5d:98 -77  6       Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:04:c3:c3:c5 -67  6       Y  FR WPA2(PSK/AES/AES)
                           Lobby e4:aa:5d:e9:dd:52 -83  1       Y  FR WPA2(802.1x/AES/AES)
                          Lounge e4:aa:5d:e9:dd:50 -83  1       Y  FR WPA2(802.1x/AES/AES)
                           Opera e4:aa:5d:e9:dd:53 -83  1       Y  FR NONE
                         Spectre e4:aa:5d:e9:dd:51 -81  1       Y  FR WPA2(802.1x,FT-802.1x/AES/AES)
                           Hobby e4:aa:5d:e9:dd:56 -82  1       Y  FR WPA2(802.1x/AES/AES)
             FunAndSeriousMobile 8a:15:04:c3:4c:45 -55  1       Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:14:c3:c3:c5 -68  40      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 8a:15:14:c3:c3:c2 -68  40      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8a:15:14:c3:c3:c1 -68  40      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:14:c4:d2:a5 -83  36      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 8a:15:14:c4:d2:a2 -84  36      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8a:15:14:c4:d2:a1 -84  36      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:14:c3:49:a5 -84  36      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 8a:15:14:c3:49:a2 -86  36      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8a:15:14:c3:49:a1 -87  36      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 9e:15:54:50:59:6e -78  48      Y  FR WPA2(PSK/AES/AES)
              The Galion Project 96:15:54:50:5d:98 -83  48      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 82:15:54:50:59:6e -78  48      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 9e:15:54:50:5d:98 -83  48      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8e:15:54:50:59:6e -75  48      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 82:15:54:50:5d:98 -83  48      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8e:15:54:50:5d:98 -81  48      Y  FR WPA2(PSK/AES/AES)
             FunAndSeriousMobile 8a:15:14:c3:4c:45 -57  44      Y  FR WPA2(PSK/AES/AES)
                FunAndSeriousSIP 8a:15:14:c3:4c:42 -57  44      Y  FR WPA2(PSK/AES/AES)
              FunAndSeriousParis 8a:15:14:c3:4c:41 -58  44      Y  FR WPA2(PSK/AES/AES)

I need to connect my MKR1000 to the "FunAndSeriousParis" wifi network but I can't. I use the ConnectWithWPA exemple sketch to do my test. It's strange because in the past it's work properly on the same wifi network.

Other strange behavior : When I run the ScanNetworksAdvanced sketch at office, I just see a small list of network, not all.
Here is the output : 

    Scanning available networks...
    ** Scan Networks **
    number of available networks: 3
    1) Signal: -74 dBm	Channel: 6		BSSID: 8A:15:04:C3:C3:C5	Encryption: WPA		SSID: FunAndSeriousMobile
    2) Signal: -74 dBm	Channel: 6		BSSID: 8A:15:04:C3:C3:C2	Encryption: WPA		SSID: FunAndSeriousSIP
    3) Signal: -74 dBm	Channel: 6		BSSID: 8A:15:04:C3:C3:CF	Encryption: WEP		SSID: 

The last network is strange : No SSID and WEP security : I can't see it with my mac.
I can connect to the "FunAndSeriousMobile" network without any problem.

When I put my phone in wifi AP mode : 
- Can't connect on it at office (don't appears on ScanNetworksAdvanced sketch)
- Work like a charm at home (appears on ScanNetworksAdvanced sketch)

I think my Arduino can't see all the wifi network available and I can't connect to it, but I don't know why.
Perhaps a bug in the WiFi101 library or the WINC1501 firmware ?

Someone have any idea ?
I'm available to perform some test or give more information.

Thanks

sandeepmistry, 2018-04-02:
Hi @hiteule,

Based on https://en.wikipedia.org/wiki/List_of_WLAN_channels - the channels above 11 are 5 GHz channels. The WINC1500 chip on the MKR1000 is **only** compatible with 2.4 GHz WiFi networks.

hiteule, 2018-04-05:
Hi,

My bad, I forgot to check this "little" thing. You're totally right !

Thanks for your help.

closed 2018-04-05 by hiteule